bambulab/BambuStudio #3598: Video cam preview makes Bambu Studio crashing.

marogme, 2024-02-15:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 64bit

### Additional system information

CPU Intel(R) Core(TM) i7-10700K
64Gb RAM
Nvida Geforce RTX 2070 Super

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1 Open Bambu Studio
2 Go to Device 
3 Click on play symbol to see live feed from camera
4 after about 40s to 1m Bambu Lab shuts down (crash)

### Actual results

cant upload anything as the result of above is that bambu studio is shut down.

### Expected results

not sure what tape here ....

### Project file & Debug log uploads

[crash_Thu_Feb_15_19_16_14_0_done.log](https://github.com/bambulab/BambuStudio/files/14301738/crash_Thu_Feb_15_19_16_14_0_done.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-02-18:
Please refer the wiki to upload crash log.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue
Please refer this wiki to see if it can be fixed by configure graphics.
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

fahaaaxiki, 2024-12-06:
Hi， we have solved this issue in our latest Public version：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50), you can update to this version to check if it works

marogme, 2024-12-06:
all good now, thank you.