bambulab/BambuStudio #3601: cut operation

3dfixer, 2024-02-16:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

p1s

### How to reproduce

load model
it contains pockets for magnets
slice
pockets still there
perform cut on model
slice and pockets disappear
perform same process on prusaslicer and it works

### Actual results

pockets disappear 

### Expected results

pockets remain

### Project file & Debug log uploads

[top case (1).gcode.zip](https://github.com/bambulab/BambuStudio/files/14315429/top.case.1.gcode.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-02-18:
I'm very sorry, I didn't understand the problem you described. Could you please provide a more detailed description, record a video, or take a screenshot to explain the problem you found? Thank you

Maximus-CZ, 2024-02-24:
@Haidiye00 
Not OP but encountered the same issue:
I have this model (pic 1).
Near the top, there is internal space for magnets (pic 2)
For testing, before I print whole model, I want to print just that piece. So in Bambu I cut the model (remove everything underneath) (pic 3)
But after cut (and slicing) the internal hole just gets filled with sparse infill, and the wall that holds the magnet is deleted completely (pic 4)
[image]
[image]
[image]
[image]

Haidiye00, 2024-02-26:
Can you provide the model file? Thank you

Maximus-CZ, 2024-02-26:
@Haidiye00 
The model I tried:
https://files.printables.com/media/prints/16623/stls/1973987_b23590d9-5cbd-468b-ac08-416fd03d220a/resource-box-4x2.stl

If that link doesnt work then its 
**Resource Box (4x2).stl**
https://www.printables.com/model/16623-terraforming-mars-storage-solution-insert/files
[image]

Haidiye00, 2024-03-06:
I have fixed this issue, please wait for version V1.9. Thank you for your discovery and support~

closed 2024-05-06 by Haidiye00