bambulab/BambuStudio #3603: Bambu Slicer renders infill settings on some settings when 999 walls should override.

vtjballeng, 2024-02-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Workstation Independent. Replicated on several machines. 

### Printer

Bambu X1C

### How to reproduce

1. [Load attached 3MF](https://github.com/bambulab/BambuStudio/files/14317963/Slicer_Failure.zip)
2. Slice Settings
3. Inspect layers. This looks good and expected for 999 walls
[image]
4. Continue to inspect layers and find some are 999 walls and some have a partial infill. 
[image]

This happens with this particular part 15deg off the recommended flat orientation in 2 axes.  If I increase the angle further or leave the part flat, slicing has the correct and expected result with 999 walls. [Here is a video showcasing the issue.](https://youtu.be/mYZ-_6zwtXQ)

### Actual results

Some layers represent the infill settings. Even if the infill is 100%, it can change the print orientation which is undesirable for some parts. 

### Expected results

999 walls should override the infill settings and lead to a consisten all walls part. 

### Project file & Debug log uploads

[Slicer_Failure.zip](https://github.com/bambulab/BambuStudio/files/14317985/Slicer_Failure.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-04-19:
It seems top one layer wall cause that. i wil check and fix it.

vtjballeng, 2024-08-09:
@QingZhangBambu , was this resolved?