bambulab/BambuStudio #3604: Pittura supporti

Adrix-64, 2024-02-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11 Pro 64 bit

### Additional system information

Intel I7

### Printer

A1 mini + ams Light

### How to reproduce

[image]
[image]
Anche dipingendo dove NON voglio supporti il software li mette comunque

### Actual results

Non cambia nulla
[image]

### Expected results

Dovrei avere i supporti solo dove non dipinto di rosso 
[image]

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-18:
@Adrix-64 could you share your 3mf with the support painting data?

thanks

Adrix-64, 2024-02-18:
Grazie a te per la risposta. Non mi lascia inserire fle che non siamo quelli evidenziati nello screenshot, come posso fare??? Grazie
[image]

lanewei120, 2024-02-18:
@Adrix-64 you can packed it into a zip file, and then upload here

Adrix-64, 2024-02-18:
Grazie, provvedo
[VERNICIATURA SUPPORTI.zip](https://github.com/bambulab/BambuStudio/files/14322044/VERNICIATURA.SUPPORTI.zip)