bambulab/BambuStudio #3606: Consistent Homing when slicer is open

mwtbn, 2024-02-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 3600x Memory: 32 GB Display Card: Nvidia RTX 2060

### Printer

Bambu Lab P1S

### How to reproduce

It might not work for you, I can only seem to produce it:
1: Slice a model
2: Select Print
3: Thats it

### Actual results

It will home at consistent times when printing, but only with Bambu Studio (or Orca Slicer) open. [Forum post](https://forum.bambulab.com/t/bambu-lab-p1s-homing-randomly/59149?u=thatbadneighbor) and [Video](https://youtu.be/b7vtWDxMKS0)

### Expected results

It should setup and print normally, without homing more than twice during setup.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14320339/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

fscottcopeland, 2024-02-20:
This isn't an isolated slicer issue. This happens to me also regardless of if I print via SD card, MakerWorld etc. 

This has also happened to another Reddit user - see post below:

https://www.reddit.com/r/BambuLab/s/5bIFpm8q12

The worst thing is it tries to home during bed levelling, and results on the nozzle dragging in the build plate, scratching it.

mwtbn, 2024-02-20:
Maybe its different from me. I ran calibration and it wont home like that. I can usually run prints from the SD card and it wont do that.

mwtbn, 2024-02-21:
I've recently got a network update on Bambu Studio and it seems to have fix my problem, I'll test it for a few days and get back about it.

mwtbn, 2024-02-26:
Welp, it started again. Same exact spots and I don't even know what i did.

walterwongbbl, 2024-03-01:
@mwtbn 
Currently, I am unable to analyze this issue from a software perspective. It seems to be related to hardware. I recommend submitting a support ticket or seeking assistance from hardware experts to address this matter.

mwtbn, 2024-03-02:
I created a ticket, so now we just have to wait for them to respond