bambulab/BambuStudio #3607: linux compatibility

dieguesmosken, 2024-02-17:
I always use the Linux operating system, I currently use the Debian 12 system, I would like compatibility to be able to install Bamboo Studio on it, I would also like you to support the Brazilian community of Bamboolab by bringing the Brazilian Portuguese version to the program.

[image]

lanewei120, 2024-02-18:
@dieguesmosken 
linux version is available on https://github.com/bambulab/BambuStudio/releases
we will refine the text on our offcial website later

dieguesmosken, 2024-02-18:
Thanks, I'll take a look and test it.

lanewei120, 2026-03-24:
> > [@dieguesmosken](https://github.com/dieguesmosken) linux version is available on https://github.com/bambulab/BambuStudio/releases we will refine the text on our offcial website later
> 
> Hi, this releases are unusable on modern Linux. What should I do? I cannot use my printer

how about using the version from https://flathub.org/apps/com.bambulab.BambuStudio