bambulab/BambuStudio #3608: MicroSD Card - Model - File Name Showing A Different File Name On Hover

ThePieMonster, 2024-02-17:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

N/A

### Printer

X1C

### How to reproduce

Navigate to MicroSD Card and then select "Model" at the top.

Hover over a model to view the full file name. 

### Actual results

The filename seems to be some cached file name. 

[image]

### Expected results

Shows the correct full file name.

[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2024-02-18:
I see, what is displayed here is the model title, not necessarily the file name.

ThePieMonster, 2024-02-18:
> I see, what is displayed here is the model title, not necessarily the file name.

Interesting. It should be the complete 3MF file name as shown right above.

Either way it is still incorrect.