bambulab/BambuStudio #3610: BambuStudio Access Violation error on i9 14900k

rd400, 2024-02-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 (latest)

### Additional system information

Intel i9 14900k
32gb RAM

### Printer

P1S

### How to reproduce

1. Open complex model
2. Click Slice

### Actual results

Program crashes with 'Access Violation'. It appears that Bambu Studio cannot handle the E-Cores on the 14900k.  If you go into task manager-->Details-->bambu-studio-->Set Affinity-->Select only CPUs 0-7. No more crashing. BambuStudio will still handle basic slicing, but nothing complex until you turn off E-cores [8-31]. Sometimes it completely blue-screens, or the computer hard locks and must be powered down.

I'm not a programmer, so unclear exact reason why this workaround is successful. 
[image]

### Expected results

Program should slice without crashing.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14320699/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-02-18:
the same issue as https://github.com/bambulab/BambuStudio/issues/3318

QingZhangBambu, 2024-02-19:
@rd400 maybe you can do some CPU tests. Some users change the CPU, all problem be solved. https://community.intel.com/t5/Processors/i9-13900K-very-frequent-crashes-Windows-11-with-apps-games-and/m-p/1528789/highlight/true
https://github.com/bambulab/BambuStudio/issues/3354

Cypher87, 2024-02-23:
I have the same issue with the i9-14900k. Whenever I slice a complex object it crashes. Mostly during generation of supports. Using compatibily mode "Windows 8" it works fine.

I tested my CPU with cinebench r23 and no issues there. Also tested my memory, tried XMP on/off. Everything else is running perfectly on this system.

 I analyzed a crashdump and it comes back with:

    KEY_VALUES_STRING: 1
    Key  : AV.Fault
    Value: Read

    Key  : Analysis.CPU.mSec
    Value: 1718

    Key  : Analysis.DebugAnalysisManager
    Value: Create

    Key  : Analysis.Elapsed.mSec
    Value: 8134

    Key  : Analysis.Init.CPU.mSec
    Value: 406

    Key  : Analysis.Init.Elapsed.mSec
    Value: 19449

    Key  : Analysis.Memory.CommitPeak.Mb
    Value: 348

    Key  : Timeline.OS.Boot.DeltaSec
    Value: 369

    Key  : Timeline.Process.Start.DeltaSec
    Value: 9

    Key  : WER.OS.Branch
    Value: ni_release

    Key  : WER.OS.Timestamp
    Value: 2022-05-06T12:50:00Z

    Key  : WER.OS.Version
    Value: 10.0.22621.1

    Key  : WER.Process.Version
    Value: 1.0.0.0

    FILE_IN_CAB:  WER.cf38a9f5-3590-4db2-8b0f-95624388d784.tmp.mdmp

    NTGLOBALFLAG:  0

    PROCESS_BAM_CURRENT_THROTTLED: 0

    PROCESS_BAM_PREVIOUS_THROTTLED: 0

    APPLICATION_VERIFIER_FLAGS:  0

    CONTEXT:  (.ecxr)
    rax=0000000000000000 rbx=0000021065db9f40 rcx=00000000ffffffff
    rdx=0000000000000000 rsi=0000009ab1ff4570 rdi=0000000000000000
    rip=00007ffe15c2d4e5 rsp=0000009ab1ff4340 rbp=0000000000bb0f2d
     r8=0000021065db9f40  r9=0000021040780e10 r10=0000021040782b20
    r11=0000021040780df0 r12=0000000000000000 r13=0000009ab1ff4950
    r14=0000000000bb0f2d r15=0000000000000000
    iopl=0         nv up ei pl zr na po nc
    cs=0033  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00010246
    BambuStudio!cereal::detail::StaticObject<cereal::detail::bind_to_archives<Slic3r::DynamicPrintConfig,cereal::detail::`anonymous namespace'::polymorphic_binding_tag> >::instantiate+0x1e5555:
    00007ffe`15c2d4e5 394d3c          cmp     dword ptr [rbp+3Ch],ecx ss:00000000`00bb0f69=????????
    Resetting default scope

    EXCEPTION_RECORD:  (.exr -1)
[... 36 lines cut ...]
    0000009a`b1ff5200 00007ffe`158e16cb     : 00000000`00000000 00000000`00000024 00000210`236a8370 00000000`00000000 : BambuStudio+0xa1807
    0000009a`b1ff5240 00007ffe`db299363     : 00000210`236a8370 00000000`00000000 00000000`00000000 00000000`00000000 : BambuStudio+0xa16cb
    0000009a`b1fff9a0 00007ffe`dd18257d     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ucrtbase!thread_start<unsigned int (__cdecl*)(void *),1>+0x93
    0000009a`b1fff9d0 00007ffe`ddb0aa58     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : kernel32!BaseThreadInitThunk+0x1d
    0000009a`b1fffa00 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x28

    SYMBOL_NAME:  BambuStudio+1e5555

    MODULE_NAME: BambuStudio

    IMAGE_NAME:  BambuStudio.dll

    STACK_COMMAND:  ~27s; .ecxr ; kb

    FAILURE_BUCKET_ID:  INVALID_POINTER_READ_c0000005_BambuStudio.dll!Unknown

    OS_VERSION:  10.0.22621.1

    BUILDLAB_STR:  ni_release

    OSPLATFORM_TYPE:  x64

    OSNAME:  Windows 10

    FAILURE_ID_HASH:  {d803ad0a-677a-62e0-ddef-19ff7e155517}

rd400, 2024-02-23:
Update: I installed a brand new i9-14900k and my Bambu Studio slicing problems disappeared. Not clear if it's a bad CPU or bad thermal paste. My error was hardware related.

closed 2024-02-23 by rd400

Arcaine10, 2024-03-18:
Good morning,
I may have found a solution without having to change the affinity each time. You have to download Intel extreme tuning utility, go to advanced tuning and set performance active-core tuning to 54. Since then I have had no more problems.
[image]