bambulab/BambuStudio #3611: Button to print files doenst show up

HashDankHog, 2024-02-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Additional system information

framework 13, 16 gb ram, ryzen 5 7640u, 152gb ssd

### Printer

Bambu x1 c

### How to reproduce

slice a file, try to print it, you cant. guessing its because The framework has a 3:2 aspect ratio

### Actual results

the button to send to printer dissapered, also text describing printer settings dissapered. 
[image]
[image]

### Expected results

for me to be able to print

### Project file & Debug log uploads

[labyrinthGiftBox_bottom.zip](https://github.com/bambulab/BambuStudio/files/14320957/labyrinthGiftBox_bottom.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bam-winterfell, 2024-03-18:
hi @HashDankHog 
We will try to reproduce this issue and find the reason.Wishing you a happy life

walterwongbbl, 2024-05-06:
@HashDankHog 
The color mapping in dark mode is incorrect, and we have made some optimizations. An update will be available soon.
We currently do not support some uncommon resolutions and aspect ratios, we are very sorry.

closed 2024-10-17 by walterwongbbl