bambulab/BambuStudio #3613: layer shift P1S after color change

Wtorres27, 2024-02-18:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

p1s

### How to reproduce

create a color change 

### Actual results

color change creates layer shift 

### Expected results

great print 

### Project file & Debug log uploads

[logs.zip](https://github.com/bambulab/BambuStudio/files/14323495/logs.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

TCC-up, 2024-02-19:
Hi,Wtorres27. Could you please submit the ticket on handy and print the time-end device logs accordingly? We need to analyze the cause of the problem together with the device logs. Thank you for your help.

zhimin-zeng-bambulab, 2024-02-19:
@Wtorres27  Could you share your gcode file?

Wtorres27, 2024-02-19:
[P1s gcode.zip](https://github.com/bambulab/BambuStudio/files/14337628/P1s.gcode.zip)
@zhimin-zeng-bambulab

algauthier, 2024-03-08:
I have resolved this issue for myself. 
I have found that completing a belt tensioning calibration by simply loosening the belt tension screws on the back of the machine, moving the print head forward and back in the middle of the printer and then leaving it in the centre-back position before tightening the screws again solves the issue completely.