bambulab/BambuStudio #3618: Mouse ear brims, PLEASE!

Sengfeng, 2024-02-19:
For those models that lift at the edges, mouse ears would be a very welcome addition.

Schokobecher, 2024-02-19:
This feature exists as a Brim Type option within OrcaSlicer, so it should be fairly easy to back port

MadMax389, 2024-02-19:
A simplified version already exists in Studio. Add a disc part.
[image]

ChrisXenon, 2024-05-16:
> A simplified version already exists in Studio. Add a disc part. [...]

Yes, but that's a very poor substitute for the real thing. I want to print 8 rectangular parts. Every corner will lift without brims. Full brims waste a lot of time and filament. Manually adding 32 circles is ridiculous.

aallen88, 2024-08-21:
Why is this still not part of BS?

I just recently got my first Bambu printer & was befuddled when I couldn't do this. I hadn't needed to until today. It's clearly been asked for and it's obviously possible. What's the hold up?

MackBambu, 2024-11-24:
Brim ear has been added.

closed 2024-11-24 by MackBambu