bambulab/BambuStudio #3619: printing error

CyThEngineeR, 2024-02-19:
### Bambu Studio Version

1.7.7.89

### Where is the application from?

Bambu Lab Official website

### OS version

windows10

### Additional system information

cpu:8600k
gpu: Asus strix 1080
ram: 32GB

### Printer

bambulab p1p

### How to reproduce

I use my normal settings for my lithophanes prints and i get error on my application that the cover fell off. when i check what is happening i notice strange print on the bed (i attach photo from the print) and the cover is down

### Actual results

i notice strange print on the bed and the cover is down
i cant sent the 3mf file because is too large is there other way to sent this ?

**please i want this to be fix as soon as possible because i have deadlines with my projects** 

### Expected results

print normal :)

### Project file & Debug log uploads

[log and photo.zip](https://github.com/bambulab/BambuStudio/files/14328567/log.and.photo.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

CyThEngineeR, 2024-02-19:
also i forget to mention i notice thick print (is on the photo i sent)

lanewei120, 2024-02-21:
@CyThEngineeR 
you can packed the 3mf to a zip file, and upload here

if it is too large, you can upload it to Dropbox or anywhere else

is it a 100% reproducable issue on your side?

XunZhangBambu, 2024-02-21:
@CyThEngineeR  Please install the cover and print again to check whether the issue persists. Thanks

XunZhangBambu, 2024-06-26:
No response for a long time, I think it can be closed

closed 2024-06-26 by XunZhangBambu