bambulab/BambuStudio #3620: Con el Ordenador y la impresora conectados en red se cae bambu studio

erikblade23, 2024-02-19:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Printer

Bambu Lab X1 Carbon + AMS

### How to reproduce

1. impresora encendida y conectada a red por wifi
2. ordenador encendido y conectado a red (da igual a LAN que a WLAN)
3. abrir bambu studio, es appimage

### Actual results

Directamente bambu studio se cierra, no puedo hacer nada.

en cambio si o bien la impresora o bien el ordenador no están en red, bambu studio inicia sin problemas

### Expected results

Deberia poder ejecutarse de forma normal permitiendo
sincronizar filamentos con AMS
visualizacion y control impresora
envio de trabajos via WLAN

### Project file & Debug log uploads

Este archivo es de cuando tengo el ordenador sin red y me deja iniciar bambu studio
[Archivos_registro_NO_problema.zip](https://github.com/bambulab/BambuStudio/files/14331995/Archivos_registro_NO_problema.zip)

Este archivo es de cuando se produce el problema por tener tanto impresora como ordenador conectados a red
[Archivos_registro_problema.zip](https://github.com/bambulab/BambuStudio/files/14331996/Archivos_registro_problema.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-02-20:
hello @erikblade23 
which appimage are you using?

and how about the flathub image from https://flathub.org/apps/com.bambulab.BambuStudio

erikblade23, 2024-02-20:
el appimage usado es:
https://github.com/bambulab/BambuStudio/releases/download/v01.08.04.51/BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage

he probado también la imagen de flathub tal y como me recomendó.

incluso he hecho una instalación nueva de Ubuntu 22.04 y tras actualizar paquetes he instalado la version de Bambu  Studio de flatpak.

el resultado es exactamente el mismo.
Si tanto la impresora como el pc estan encendidos y conectados a red bambu studio se cae

lanewei120, 2024-02-21:
could you use gdb to have a try?
#gdb ./[BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage](https://github.com/bambulab/BambuStudio/releases/download/v01.08.04.51/BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage)
#run
...
when it crashes, use 'bt' to check the stack
#bt

and show the result here

erikblade23, 2024-02-21:
thanks for your timea and sorry about my english i'll try use it

this is the #bt result

#0  0x00007fffd878a9a7 in TransCodePartial () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#1  0x00007fffd879003d in iotc_SendMessage () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#2  0x00007fffd879368f in _IOTC_SendQuryDevice5 () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#3  0x00007fffd87a4e54 in AddUDPQueryDeviceInfoTask () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#4  0x00007fffd87aad1c in IOTC_Connect_UDP_Inner () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#5  0x00007fffd87ab4a8 in IOTC_Connect_ByUIDEx () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#6  0x00007fffd87579f8 in BambuTunnelTutk::open() () from /home/ibai/.config/BambuStudio/plugins/libBambuSource.so
#7  0x00007fffd925dcc1 in BBL::TunnelMqtt::reconnect(std::unique_lock<std::mutex>&, int) ()
   from /home/ibai/.config/BambuStudio/plugins/libbambu_networking.so
#8  0x00007fffd9260b6c in BBL::TunnelMqtt::run() () from /home/ibai/.config/BambuStudio/plugins/libbambu_networking.so
#9  0x00007fffd926132a in std::thread::_State_impl<std::thread::_Invoker<std::tuple<BBL::TunnelMqtt::attached()::{lambda()#1}> > >::_M_run() ()
   from /home/ibai/.config/BambuStudio/plugins/libbambu_networking.so
#10 0x00007ffff7cdc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#11 0x00007fffefe94ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#12 0x00007fffeff26850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

and this is the archive with all shows the terminal since use gdb
[gdb_bambu.txt](https://github.com/bambulab/BambuStudio/files/14356474/gdb_bambu.txt)

if you want i can try gdb with flatpak image too

lanewei120, 2024-02-21:
@erikblade23 

thanks, no need to tryu the flatpak image
could you also use gdb to try an older version https://github.com/bambulab/BambuStudio/releases/download/v01.08.02.56/BambuStudio_linux_ubuntu_v01.08.02.56-20231228.AppImage?

then give the stack here

bambu123, 2024-02-21:
@erikblade23 

Are you willing to provide additional logs for the network module?
These logs are not output by default. You need to create a "tutk_log" subdirectory under the log directory to output them. These logs will include your device IP. It is recommended to send them to me via email.

$ mkdir ~/.config/BambuStudio/log/tutk_log

When you create the "tutk_log" directory, you need to restart Studio and reproduce the problem.

erikblade23, 2024-02-22:
Tengo sospecha que pueda tratarse de un problema del Network plugin del bambú studio, ya que el problema es común y en el mismo punto en las dos distribuciones (AppImage y flatpak) y en diferentes versiones. 
Lo que si puedo añadir es que si inicio el programa con la impresora apagada y una vez que el programa ha iniciado enciendo la impresora, no se cae y trabaja con normalidad, pero en el momento que doy a la pestaña "dispositivo" oigo como la impresora hace amago de ejecutar un movimiento y seguido el BambuStudio se cae.

bambu123, 2024-02-22:
> Tengo sospecha que pueda tratarse de un problema del Network plugin del bambú studio, ya que el problema es común y en el mismo punto en las dos distribuciones (AppImage y flatpak) y en diferentes [...]

Yes. but it is the first time I see this crash stack.

erikblade23, 2024-02-23:
Yo más que hacer instalación de Ubuntu desde cero y reinstalar el Bambú Studio no sé qué más puedo hacer. Si el culpable es el plugin (como así parece) no sé qué prueba o que acción o acciones podría hacer para intentar solucionar o al menos aproximar el problema.

bambu123, 2024-02-26:
> Yo más que hacer instalación de Ubuntu desde cero y reinstalar el Bambú Studio no sé qué más puedo hacer. Si el culpable es el plugin (como así parece) no sé qué prueba o que acción o acciones podría hacer para intentar solucionar o al menos aproximar el problema.

Please provider the tutk_log. see above.
https://github.com/bambulab/BambuStudio/issues/3620#issuecomment-1956290936

erikblade23, 2024-02-26:
Supongo necesitaré de algún canal privado para hacerlo, no?

bambu123, 2024-02-26:
> Supongo necesitaré de algún canal privado para hacerlo, no?

my email address: chunmao.guo@bambulab.com

erikblade23, 2024-02-26:
Y le he mandado the tutk_log.

tiene dos correos uno abriendo el Bambu Studio con la impresora encendida, el programa se cae directamente.
el otro es abriendo el Bambu Studio con la impresora apagada, el programa se mantiene. enciendo la impresora y es cuando voy a la pestaña dispositivo cuando el programa se cierra

bambu123, 2024-08-08:
Sorry for the long time no reply. Does this problem still exist in the latest version?