bambulab/BambuStudio #3626: SD card

okolsky, 2024-02-20:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS 14.2.1

### Printer

Bambulab P1S

### How to reproduce

1. Start printing
2. got the freeze of printing head
the 
[image]
[image]

### Actual results

printing head is freezing during the printing in different timelines. 
the desktop version can show: error of connection with printer or with server or SD card error. 
The ios app: status of the printer offline and no cantions can be possible

after ejecting and inserting of the card the process can proceed. But during this time the place is cold and the details are not glued. And on the position of the frozen head we have a lot of filament - so even the process can be somehow unfrozen - there are no needs as the details are broken. 

### Expected results

I can put the printing and go away fo ra hours...

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14342650/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

okolsky, 2024-02-20:
hello, I've changed the SD card from original one to Kingstone, and the issue - left for a couple of prints, but I have new one instead:
[image]

sodachen22, 2024-02-21:
> hello, I've changed the SD card from original one to Kingstone, and the issue - left for a couple of prints, but I have new one instead: <img alt="0300" width="432" src="https://private-user-images.githubusercontent.com/160593754/306379423-b8d9362b-080d-47cf-b20a-6034f5ef7642.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0OTU4NTIsIm5iZiI6MTcwODQ5NTU1MiwicGF0aCI6Ii8xNjA1OTM3NTQvMzA2Mzc5NDIzLWI4ZDkzNjJiLTA4MGQtNDdjZi1iMjBhLTYwMzRmNWVmNzY0Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQwNjA1NTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00OWViOThlNWUzMTdiMzk0OWY1OTVlNWEzZTc3NjdkNGVhYjVhNjQ1NGY5Y2Y1NDc4NTFlZmE4OTNhZjViOTk2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Qp5sDeGw35U3_lwdMzBKRtJW0eR-KJVLuGVxa0qLlyE">
1. This pop-up window indicates that the print job has been cancelled.

sodachen22, 2024-02-21:
2. What is the version of the printer? In the latest firmware, there are more error message when there is problem in reading/writing SD card. Is there any error messages when the problem is caused by SD card?
3. If this issue is related to printer hardware, we may need printer LOG.

sodachen22, 2024-02-21:
https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

okolsky, 2024-02-21:
Please take a look on all info above. Log, and the rest were attached.

sodachen22, 2024-02-21:
This is Studio LOG, if the issue is stopping in the middle of a printing job, we need printer LOG.

okolsky, 2024-02-24:
[log.zip](https://github.com/bambulab/BambuStudio/files/14392214/log.zip)
take a look, please, is it that one?

okolsky, 2024-02-26:
any comments, guys?

okolsky, 2024-02-29:
@sodachen22 , @lanewei120  guys, please check the logs, could you?

sodachen22, 2024-03-12:
It is suspected the issue is related to the SD card. Would you please have a try with another SD card.
If the issue is still there, please have a ticket with the printer LOG. https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log

lanewei120, 2024-03-18:
> I've provided the printer log 3 messages above, could you check?

hello

@sodachen22 is requesting printer log, not bambu stduio log
please upload the printer log following  https://wiki.bambulab.com/en/p1/troubleshooting/how-to-upload-log