bambulab/BambuStudio #3627: Flash frequently when playing live camera

NYNUZEO, 2024-02-20:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOs 14.1 (23B74)

### Additional system information

MacBook Pro
16-Inch, 2019
处理器
2.6GHz六核Intel Core i7
图形卡 AMD Radeon Pro5300M4GB
Intel UHD Graphics 630 1536 ME
内存32GB2667MHz DDR4
启动磁盘 Macintosh HD
macOS 14.1(23B74)

### Printer

Bambu a1mini

### How to reproduce

1. open the app 'bambu studio'
2. click the device tab
3. click play button

### Actual results

Image interleaved rollback

### Expected results

regular & real time & no flicker play the live camera video

### Project file & Debug log uploads

[debug_Tue_Feb_20_18_48_52_94522.log](https://github.com/bambulab/BambuStudio/files/14343258/debug_Tue_Feb_20_18_48_52_94522.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-02-21:
Your network plugin version is 01.08.04.01. Please update plugin following the tip shown in 3d plater.

bambu123, 2024-03-01:
@NYNUZEO 请问你是否解决了问题。May I ask if you have solved the problem?