bambulab/BambuStudio #3632: Network plugin is downloaded but not loaded

Vonsyth, 2024-02-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 22.04

### Additional system information

Raspberry Pi 4

### Printer

Bambu Lab P1S

### How to reproduce

1. Start Program (Flatpak) `flatpak run com.bambulab.BambuStudio`
2. Download Network Plugin

### Actual results

Plugin is downloaded, but not loaded. Request to download plugin persists.
Output in Terminal: 
`loading network module at /home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so`
`error, dlerror is /home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory`
`after dlopen, network_module is (nil)`

### Expected results

Plugin is installed and loaded properly.

### Project file & Debug log uploads

Log file: [debug_Wed_Feb_21_09_48_12_4.log.zip](https://github.com/bambulab/BambuStudio/files/14357201/debug_Wed_Feb_21_09_48_12_4.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-02-22:
is the file /home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so already there?
and is the file permission correct?

Vonsyth, 2024-02-22:
Yes, the file is at this location. I also tried to manually download the plugin and unzip the files there, Got the same result. 
[image]
These are the permissions set. Even if I change them, the plugin gets redownloaded on startup, which changes the permissions again.

lanewei120, 2024-02-22:
is it possible to write a simple program to verify whether it is accessable on your system:
just like
`
    std::string library = "/home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so";
    void* netwoking_module = NULL
    netwoking_module = dlopen( library.c_str(), RTLD_LAZY);
    if (!netwoking_module) {
        
        //netwoking_module = dlopen( library.c_str(), RTLD_LAZY);
        char* dll_error = dlerror();
        printf("error, dlerror is %s\n", dll_error);
    }
`

lanewei120, 2024-02-22:
[test_dlopen.zip](https://github.com/bambulab/BambuStudio/files/14370642/test_dlopen.zip)
you can try the binary attached:

after extract the zip file,
$chmod a+x test_dlopen
$./test_dlopen
please input the library path:
**/home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so**
opening library /home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so

Vonsyth, 2024-02-22:
The binary didn't work, since I am on a Raspberry Pi (aarch64) and the binary is for x86 systems I suppose. I compiled the code from above and got following result:
`error, dlerror is /home/pifor3d/.var/app/com.bambulab.BambuStudio/config/BambuStudioInternal/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory`
It's the same error message from above.

Vonsyth, 2024-02-22:
Ok, that explains the problem. Is there any plan in the future to make it work on aarch64-systems? The program itself is already working on aarch64 after all.

lanewei120, 2024-02-22:
I am afraid currently we don't have such plan

Vonsyth, 2024-02-22:
Is it possible for me to get the source code so I can try to compile the plugins myself?

closed 2024-02-22 by Vonsyth