bambulab/BambuStudio #3634: Changing BBL plate SVG does not update in Bambu Studio

ptsao88, 2024-02-21:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab X1C

### How to reproduce

Go to C: Program Files --> Bambu Studio --> resources --> profiles --> BBL
Replace "bbl-3dp-logo.svg" with a custom SVG
Refresh and restart Bambu Studio

### Actual results

There is no change to the displayed plate. If I do this on the "Prusa" profile and replace "mk3.svg", it works as intended. See image below.

[image]

### Expected results

I expect this to work the same as when replacing "mk3.svg" on the Prusa profile.
Perhaps there is a different SVG file that I should be replacing or modifying to add custom changes to the shown plates in Bambu Studio?

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ptsao88, 2024-02-21:
I also noticed that this feature that's available in Prusa Slicer does not work.

In Bambu Studio all the buttons are disabled and greyed out.

[image]

ptsao88, 2024-02-21:
The texture feature shows up if I create a copy of an existing printer profile. However, it doesn't seem to do anything after uploading a file.

If I create a new printer profile totally from scratch this works...but is not ideal.

[image]
[image]

Haidiye00, 2024-03-01:
Bambu's official printer does not support importing SVG hot bed patterns, only third-party printers can support importing SVG hot bed patterns

ptsao88, 2024-03-01:
That's unfortunate. I am creating customized build plates with logos in specific areas. Importing the SVG into Bambu Studio would help me align printed parts onto specific areas of the build plate to match up with the logos.

Haidiye00, 2024-03-04:
Can you draw some diagrams to illustrate your application scenario? We will evaluate and consider whether to develop it. Thank you

ptsao88, 2024-03-04:
Hi @Haidiye00 - see this video for what I would like to be able to do: https://youtu.be/TEeMEJMefow?si=dATzBYxxgM5LSls6&t=58

This would let users line up prints exactly to where they want on customized PEI plates that I am making. See example picture below.

[image]

Haidiye00, 2024-03-05:
Understood, we will evaluate and consider whether to develop it soon. Thank you

ptsao88, 2024-03-05:
Great - thank you!

Haidiye00, 2024-06-24:
Hello, can you provide your SVG? Thank you

emberprototypes, 2024-06-24:
Here is an example.

If we can import this into Bambu Studio then it will make lining up parts to the artwork easier for our custom plates that imprint logos onto the first layer.

[image]

Haidiye00, 2024-06-25:
[image] I have achieved an effect, and it would be even better if you could provide some SVGs to facilitate our internal testing

emberprototypes, 2024-06-25:
> ![image](https://private-user-images.githubusercontent.com/132528822/342544365-87eaa865-dbc1-4504-9122-eefc93715e6c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Did the SVG in the previous response not work? I can provide more if needed

Haidiye00, 2024-06-25:
If possible, could you please provide some? Thank you

emberprototypes, 2024-06-25:
> Here is an example.
> 
> If we can load these types of SVGs into Bambu Studio as the plate texture then it will make lining up parts to the artwork easier for our custom plates that imprint logos onto the first layer.
> 
> ![Bambulab+Plate_V1_out](https://private-user-images.githubusercontent.com/126627300/342433480-1cf9416e-cecd-4752-8422-066955dd68cb.svg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkyODEzNjEsIm5iZiI6MTcxOTI4MTA2MSwicGF0aCI6Ii8xMjY2MjczMDAvMzQyNDMzNDgwLTFjZjk0MTZlLWNlY2QtNDc1Mi04NDIyLTA2Njk1NWRkNjhjYi5zdmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyNVQwMjA0MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MjhlMTQzYThhNGJlN2NmZGRjNWUwZTM2YzI1NjUxZmQxM2ZkODZiYzYyZmJmNWE3NDJjMDIzM2EzNTYwZWVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jFCeMP0MfGxHcw2gszia0A7rOxDVpbWJ69erDM8fAKo)

I attached one in this reply...

Haidiye00, 2024-06-25:
Can you provide the original files

emberprototypes, 2024-06-25:
[image]
[image]
[image]
[image]
[image]

See attached for examples.

emberprototypes, 2024-06-25:
> Can you provide the original files

You should be able to just right click --> save as SVG.

Haidiye00, 2024-06-25:
[image]
I have tested your file and the effect is as shown above. I suggest that your file does not have a black background, as the second size does not seem to match exactly.

emberprototypes, 2024-06-25:
Are you sure that the SVG is not being scaled funny when you load it as a texture?

You can see that the SVG should have the correct dimensions for the plate.

Also - if you load the SVG into Bambu Studio, you can see it converts it to a 3D model that is the correct size.

[image]
[image]

Haidiye00, 2024-06-26:
[image]
The problem has been resolved. Take a look at the effect.

emberprototypes, 2024-06-26:
Yes this is perfect!!! Will this be on the next Bambu Studio release? 😀

Haidiye00, 2024-06-27:
yes

emberprototypes, 2024-06-27:
Amazing thank you so much!

Haidiye00, 2024-06-27:
You're welcome, but the next version may take quite some time. Perhaps in about a quarter~

emberprototypes, 2024-06-27:
Ok no problem! I'm hoping the next version also comes with filament specific XY shrinkage compensation 😬

Thank you for your hard work.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00

emberprototypes, 2024-10-16:
Amazing thank you @Haidiye00 !