bambulab/BambuStudio #3635: Linux Studio dropping events (mouse, keyboard, right click, menu) resulting in degraded functionality and performance

gpeden, 2024-02-21:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21.3 Cinnamon

### Additional system information

System:
  Kernel: 5.15.0-94-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Micro-Star product: MS-7C56 v: 1.0 serial: <superuser required>
  Mobo: Micro-Star model: MPG B550 GAMING PLUS (MS-7C56) v: 1.0 serial: <superuser required>
    UEFI: American Megatrends LLC. v: 1.F0 date: 10/11/2023
Battery:
  Device-1: hidpp_battery_0 model: Logitech MX Keys Wireless Keyboard serial: <filter>
    charge: 100% (should be ignored) rechargeable: yes status: Full
CPU:
  Info: 8-core model: AMD Ryzen 7 5800X bits: 64 type: MT MCP smt: enabled arch: Zen 3 rev: 0
    cache: L1: 512 KiB L2: 4 MiB L3: 32 MiB
  Speed (MHz): avg: 2372 high: 4748 min/max: 2200/3800 boost: enabled cores: 1: 2200 2: 2199
    3: 2200 4: 2200 5: 2199 6: 2200 7: 2200 8: 2200 9: 2424 10: 4748 11: 2199 12: 2200 13: 2200
    14: 2199 15: 2199 16: 2199 bogomips: 121595
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: NVIDIA GM206 [GeForce GTX 950] vendor: PNY driver: N/A pcie: speed: 2.5 GT/s lanes: 16
    bus-ID: 2b:00.0 chip-ID: 10de:1402 class-ID: 0300
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nouveau,vesa
    unloaded: fbdev,modesetting gpu: N/A display-ID: :0 screens: 1
  Screen-1: 0 s-res: 2560x1440 s-dpi: 96 s-size: 677x381mm (26.7x15.0") s-diag: 777mm (30.6")
  Monitor-1: default res: 2560x1440 hz: 93 size: N/A
  OpenGL: renderer: llvmpipe (LLVM 15.0.7 256 bits) v: 4.5 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes
Audio:
  Device-1: NVIDIA GM206 High Definition Audio vendor: PNY driver: snd_hda_intel v: kernel pcie:
    speed: 2.5 GT/s lanes: 16 bus-ID: 2b:00.1 chip-ID: 10de:0fba class-ID: 0403
  Device-2: AMD Starship/Matisse HD Audio vendor: Micro-Star MSI driver: snd_hda_intel v: kernel
    pcie: speed: 16 GT/s lanes: 16 bus-ID: 2d:00.4 chip-ID: 1022:1487 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-94-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Micro-Star MSI
    driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 2a:00.0
    chip-ID: 10ec:8168 class-ID: 0200
  IF: enp42s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM20702A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-6:5
    chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 220e hci-v: 4.0 rev: 1000
Drives:
  Local Storage: total: 1.95 TiB used: 587.12 GiB (29.5%)
  ID-1: /dev/nvme0n1 vendor: Western Digital model: WDBRPG5000ANC-WRSN size: 465.76 GiB
    speed: 31.6 Gb/s lanes: 4 type: SSD serial: <filter> rev: 111130WD temp: 38.9 C scheme: GPT
  ID-2: /dev/sda vendor: Western Digital model: WD6400AAKS-22A7B2 size: 596.17 GiB
    speed: 3.0 Gb/s type: N/A serial: <filter> rev: 3B01
  ID-3: /dev/sdb vendor: Western Digital model: WD10EZEX-00BN5A0 size: 931.51 GiB
    speed: 6.0 Gb/s type: HDD rpm: 7200 serial: <filter> rev: 1A01 scheme: MBR
Partition:
  ID-1: / size: 365.24 GiB used: 39.51 GiB (10.8%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 190.5 MiB used: 6.1 MiB (3.2%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 915.82 GiB used: 272.53 GiB (29.8%) fs: ext4 dev: /dev/sdb1
Swap:
  ID-1: swap-1 type: partition size: 93.13 GiB used: 2.55 GiB (2.7%) priority: -2
    dev: /dev/nvme0n1p3
USB:
  Hub-1: 1-0:1 info: Hi-speed hub with single TT ports: 10 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-2: 1-2:2 info: Genesys Logic Hub ports: 4 rev: 2.0 speed: 480 Mb/s power: 100mA
    chip-ID: 05e3:0608 class-ID: 0900
  Device-1: 1-2.3:4 info: SanDisk SDDR-B531 type: Mass Storage driver: usb-storage interfaces: 1
    rev: 2.1 speed: 480 Mb/s power: 500mA chip-ID: 0781:cfcb class-ID: 0806 serial: <filter>
  Device-2: 1-2.4:6 info: Yubicom Yubikey 4/5 OTP+U2F+CCID type: Keyboard,HID,Smart Card
    driver: hid-generic,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 30mA chip-ID: 1050:0407
    class-ID: 0b00
  Device-3: 1-5:3 info: Logitech Unifying Receiver type: Keyboard,Mouse,HID
    driver: logitech-djreceiver,usbhid interfaces: 3 rev: 2.0 speed: 12 Mb/s power: 98mA
    chip-ID: 046d:c52b class-ID: 0300
  Device-4: 1-6:5 info: Broadcom BCM20702A0 Bluetooth 4.0 type: Bluetooth driver: btusb
    interfaces: 4 rev: 2.0 speed: 12 Mb/s chip-ID: 0a5c:21e8 class-ID: fe01 serial: <filter>
  Device-5: 1-7:7 info: Micro Star MYSTIC LIGHT type: HID driver: hid-generic,usbhid
    interfaces: 1 rev: 1.1 speed: 12 Mb/s power: 500mA chip-ID: 1462:7c56 class-ID: 0300
    serial: <filter>
  Device-6: 1-10:8 info: Kensington Expert Wireless Trackball Mouse (K72359WW) type: Mouse,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 12 Mb/s power: 100mA chip-ID: 047d:8018
    class-ID: 0300
  Hub-3: 2-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
  Hub-4: 3-0:1 info: Hi-speed hub with single TT ports: 4 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-5: 4-0:1 info: Super-speed hub ports: 4 rev: 3.1 speed: 10 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  Message: No sensor data found. Is lm-sensors configured?
Repos:
  Packages: 3453 apt: 3421 flatpak: 9 snap: 23
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/1password.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/1password-archive-keyring.gpg] https: //downloads.1password.com/linux/debian/amd64 stable main
  Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list
    1: deb [signed-by=/usr/share/keyrings/brave-browser-archive-keyring.gpg arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
  Active apt repos in: /etc/apt/sources.list.d/github-cli.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/githubcli-archive-keyring.gpg] https: //cli.github.com/packages stable main
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/nordvpn.list
    1: deb https: //repo.nordvpn.com/deb/nordvpn/debian stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  No active apt repos in: /etc/apt/sources.list.d/playonlinux.list
  No active apt repos in: /etc/apt/sources.list.d/steam-beta.list
  Active apt repos in: /etc/apt/sources.list.d/steam-stable.list
    1: deb [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
    2: deb-src [arch=amd64,i386 signed-by=/usr/share/keyrings/steam.gpg] https: //repo.steampowered.com/steam/ stable steam
Info:
  Processes: 454 Uptime: 1d 15h 15m wakeups: 2 Memory: 62.72 GiB used: 14.34 GiB (22.9%)
  Init: systemd v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 Client: Cinnamon v: 6.0.4
  inxi: 3.3.13

### Printer

Bambu Lab X1C

### How to reproduce

Use Studio for a while.  Notice events start getting missed and program stop working correctly.  Keypresses, mouse clicks, right click context menus, etc. etc. etc.

### Actual results

Use Studio for a while.  Notice events start getting missed and program stop working correctly.  Keypresses, mouse clicks, right click context menus, close program, etc. etc. etc. etc. etc. etc. I finally had to 'end' from Process Info.

Upon restart the program seems to work correctly / normally - until it doesn't.

### Expected results

Events should not be missed and the program should function normally.

### Project file & Debug log uploads

project doesn't matter. log files attached.
[log.zip](https://github.com/bambulab/BambuStudio/files/14364081/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-02-22:
hello @gpeden 
how long will bambu studio run before this issue appear?
and what have been done on bambu studio?

furthermore, could you help to use 'top' to check the cpu usuage when this issue happens?

gpeden, 2024-02-23:
Seems like it takes about a day or so for it to start dropping input events.  sorry its not more consistent.

lanewei120, 2024-02-26:
we need to find a way ro reproduce it
1. what have been done on bambu studio?
2. could you give a snapshot of top when bambu studio starts normally, and when it lost events