bambulab/BambuStudio #3638: MODIFIERS leave lines in model.

MrJawz, 2024-02-23:
### Bambu Studio Version

1.8.4.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

11

### Additional system information

amd ryzen 9

### Printer

x1c

### How to reproduce

Add modifier to model. modifier cube for example. Then save model as a project so you can edit the file later incase you need to make a change. reopen the file and make a change to the model. Slice it and then you'll see lines where the modifier is at. Thats the best way to describe it. I'll add a picture
[image]
[image]
[image]
[image]

### Actual results

The end result is lines that show on the finished product where modifiers were added
[image]

### Expected results

not sure what to say

### Project file & Debug log uploads

dont know how to do this

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-27:
@MrJawz 
thanks for reporting this issue
to help us better understand your case

could you upload your 3mf file as a zip here?

TeDsWeb, 2024-02-27:
I think the problem is that an outer wall is created inside the object and both the modified object and the original object are treated internally as two different objects that are merged together. I have made a simple example with a 3DBenchy to demonstrate this effect.
[3DBenchy_Modifier.zip](https://github.com/bambulab/BambuStudio/files/14424246/3DBenchy_Modifier.zip)

MrJawz, 2024-02-27:
Hi, So what is the fix for this to prevent an outer wall effect to occur?

TeDsWeb, 2024-02-27:
I'm just another user with the same problem^^ and I'm sorry to say this, but I don't think there's a solution at the moment, and even Prusaslicer has the same problem if you adding more walls to a modified section. We can only hope that someone from bambu labs fixes this by only treating the inside walls differently for a modified section and leaving the outside walls as they are, except for fuzziness or something.

eried, 2024-02-28:
I also noticed this, but also I think it is a difficult thing to solve, the shell should be considered as an "untouchable" thing

MrJawz, 2024-02-28:
> I agree with you.

Thanks guys I’m glad I’m not the only one experiencing this problem. I hope
bambu fixes it

ScottyMakesStuff, 2024-11-10:
I have the same problem.
I was attempting to strengthen a thin walled model at a hinge joint and the hinges all cracked along the modifier edge line.
Surely there's some way to make the two outer walls along the boundary overlap enough to prevent this place of weakness