bambulab/BambuStudio #3640: Flushing volume only shows 2 digits

gpeden, 2024-02-23:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21.3 Cinnamon

### Additional system information

[image]

### Printer

Bambu Lab X1C

### How to reproduce

1. Open Flushing Volumes window
2. Enter 3 digit flushing volume (I need 400 for example)
3. Flushing volume shows '00' - looks like text box is too small

### Actual results

3. Flushing volume shows '00' - looks like text boxes are too small
[image]

### Expected results

3. Flushing volume should show value entered, in my case '400'

### Project file & Debug log uploads

[Flushing_Volumes.zip](https://github.com/bambulab/BambuStudio/files/14388352/Flushing_Volumes.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

nadnerb33, 2024-05-22:
I have the same issue with PopOS (Ubuntu) 20.04 and every version of Bambu Studio I have tried (appimage, flatpak etc) including the latest flatpak version 1.9.1.66

Changing the OS text size has some influence, but the boxes are still too small.

Default text size
[image]

Large text size:
[image]

Ralf-3D, 2024-06-08:
I have the same problem.

BambuStudio AppImage version 1.9.1.66.

LinuxMint 21.3 xfce see folowing system-info:
System:
  Kernel: 5.15.0-112-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Xfce 4.18.1
    tk: Gtk 3.24.33 info: xfce4-panel wm: xfwm 4.18.0 vt: 7 dm: LightDM 1.30.0
    Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy

This bug also occures in OrcaSlicer OrcaSlicer 2.0.0 AppImage

[image]

MklBambu, 2024-06-13:
Sorry, it is a issue with wxWidgets.
we will do our best to solve it

KeluDev-Code, 2025-02-10:
Hi, I have the same problem.
Is there any update on this issue?
Maybe a minimum width could be set

andruhon, 2025-03-19:
Same issue 
Arch with KDE
[image]
Tried to force the popup to fullscreen, but it didn't help

[image]

andruhon, 2025-03-28:
Looks better in version 2.0.0.95

[image]

XunZhangBambu, 2025-03-28:
@andruhon @KeluDev-Code @longrackslabs @nadnerb33 @Ralf-3D 
In the latest version, we are using a new approach to display this interface, and this should resolve your issue.

longrackslabs, 2025-03-28:
Yes, I installed 2.0 and the new UI for this looks much better now.

I'm happy to close this issue.

closed 2025-09-22 by longrackslabs