bambulab/BambuStudio #3641: Multi-material + Height range modifier triggers "The prime tower is only supported if all objects have the same variable layer height"

saladpanda, 2024-02-24:
### Bambu Studio Version

1.8.3.89

### Where is the application from?

Bambu Lab github releases

### OS version

Arch Linux

### Printer

Bambu Lab P1S

### How to reproduce

I tried stackprinting gridfinity baseplates and alternating PLA and PETG for easy separation. I created the stack but when I added a height range modifier to lower the speed of certain layers I could not slice as I got the error
"Error: The prime tower is only supported if all objects have the same variable layer height"

This is the minimal example I came up with that triggers the error (also see 3mf file below)
[image]

### Actual results

Can not slice because of
"Error: The prime tower is only supported if all objects have the same variable layer height".

### Expected results

Should slice without errors as no variable layer height is used and layer height is consistently 0.2mm across the whole project.

### Project file & Debug log uploads

Original Project file for stack printing gridfinity baseplates: 
[gridfinity-clip-base-remix 5x5 stack8 _ primetower error.zip](https://github.com/bambulab/BambuStudio/files/14392234/gridfinity-clip-base-remix.5x5.stack8._.primetower.error.zip)

minimal example triggering the error:
[primetower error - minimal example.zip](https://github.com/bambulab/BambuStudio/files/14392236/primetower.error.-.minimal.example.zip)

Very similar error in PrusaSlicer that has been resolved long ago:
https://github.com/prusa3d/PrusaSlicer/issues/7711

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-02-27:
Thanks, we will fix it in the next version.

zhimin-zeng-bambulab, 2024-03-27:
It has been fixed in v01.09.00.06

closed 2024-03-27 by zhimin-zeng-bambulab

adminmat, 2024-07-22:
Orca 2.1.1. I still have this issue. Wont allow me to slice although I've synchronized the two objects to have the same layer heights. 

[image]