bambulab/BambuStudio #3643: P1P打印直播流点击播放按钮，重复弹出需重新注册BambuSource的错误窗口

countree103103, 2024-02-25:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 专业工作站版

### Printer

Bambu Lab P1P

### How to reproduce

点击播放按钮出现
[image]
点击Yes后，出现
[image]

### Actual results

在上述过程后，点击播放按钮任无法播放。再次点击播放按钮会重复出现上述情况。
错误信息为
[image]

### Expected results

直播流正常播放。

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14395083/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-02-27:
可能注册被杀毒软件干扰了。参照wiki 手动注册。检查注册表。 https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

bambu123, 2024-03-01:
@countree103103 请问你是否解决了问题。

countree103103, 2024-04-04:
> @countree103103 请问你是否解决了问题。

我尝试了wiki里的手动导入reg，没有解决。
以下是reg内容：

Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\Software\Classes\CLSID\{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}]
@="BambuSource Filter"

[HKEY_CURRENT_USER\Software\Classes\CLSID\{233E64FB-2041-4A6C-AFAB-FF9BCF83E7AA}\InprocServer32]
@="C:\\Users\\71766.ChenZY-Win11\\AppData\\Roaming\\BambuStudio\\plugins\\BambuSource.dll"
"ThreadingModel"="Both"

bambu123, 2024-04-07:
还有另一个注册项目，再试试

https://wiki.bambulab.com/software/bambu-studio/live-view/bambu.reg

countree103103, 2024-04-30:
> 还有另一个注册项目，再试试
> 
> https://wiki.bambulab.com/software/bambu-studio/live-view/bambu.reg

注册了这个reg之后可以加载视频流了。
之前的wiki里没有这个bambu.reg的说明，可以加一下说明。

closed 2024-04-30 by countree103103