bambulab/BambuStudio #3644: Wall order that can be changed by a modifier / ability to change wall order for only the first layer

berri3D, 2024-02-25:
**Is your feature request related to a problem? Please describe.**
For certain models, I would like to be able to change the wall order for the first layer only and not the rest of the model.

Models with small first layer details often benefit from the outer wall being laid first so that the inner wall has something to stick to. However, above the first layer it's better to have the outer wall printed last so changing it for the entire model doesn't make sense.

Here is an example where small inner walls are a problem:

[image]

Those little circles get dragged by the nozzle into other areas of the print.  Here is what the rest of the layer looks like:

[image]

If the outer walls were printed first, then that would eliminate the problem.

However, you cannot change just the first layer wall order, even with a modifier.  I used to be able to do it with modifiers, but I'm either mistaken that I could or it's missing from the menu when it ought not to be. At any rate, I know Orca Slicer can do it. Here is a comparison between it and Studio:

[image]

Orca's quality menu is what I thought Studio's menu looked like before. FYI, developer mode was been turned on before taking these screenshots.

**Describe the solution you'd like**
I would like the ability to target just the first layer's wall order either with a one-click option (preferred) or by using a modifier.

**Describe alternatives you've considered**
A dirty bed, an insufficiently heated bed, and/or wet filament can cause adhesion issues. I have eliminated these as being factors for the inner walls being dragged. I've also slowed down the first layer speed. Decreasing the travel speed for the first layer would also be an idea, but it's not an option in Studio (it is in Orca).

The long and short of it is that extra small details don't a;ways have sufficient bonding power on their own even with a pristine bed, etc. The smaller details there are, the more likely they will get dragged. Increasing the size of the model so as to make the inner layers bigger isn't an option because the model is designed to be the size that it is.

**Additional context**
Add any other context or screenshots about the feature request here.

toddanglin, 2025-04-07:
Big +1. I would also like to see these options in Studio's modifier properties.

I have some models that print best when the first few layers printed use a different wall order than the rest of the print. Generally, I use Orca Slicer to achieve this result, but with the H2D I'm re-evaluating Studio for my workflow. Unfortunately, this is a critical blocker to making the switch from Orca.

As a general statement, it would be great to see nearly all properties available for the modifiers when using "Advanced Mode." Limit to the most common properties when not using Advanced mode.

mccannex, 2025-04-24:
+1 yes please. Tiny dots on the first layer just get dragged everywhere.