bambulab/BambuStudio #3645: 在没网时切换到"设备"页程序失去响应

ChenKe404, 2024-02-25:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

CPU: 15-12600kf
MEM: 32GB
GPU: 3070ti

### Printer

Bambu Lab A1 mini

### How to reproduce

连接wifi但断开互联网 -> 打开Bambu Studio -> 切换到"设备"标签页

### Actual results

换到“设备“标签页后程序就会卡顿甚至无响应，连程序窗口都无法拖动。
[image]

### Expected results

能让我正常从设备列表选择我局域网里的打印机。

### Project file & Debug log uploads

不需要项目文件
[log.zip](https://github.com/bambulab/BambuStudio/files/14396426/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-03-01:
我们的软件必须依赖互联网才能工作。 除非打印机是lan only模式。

walterwongbbl, 2024-03-01:
这个现象每次都会出现吗。 我们必须依赖互联网执行我们的一些网络请求操作。

ChenKe404, 2024-03-01:
@walterwongbbl 这个情况每次都会出现，无论我的机器是否开启局域网模式。有时用机地点网络不正常，我仍然希望在没有网络服务的情况下正常使用切片软件控制机器进行生产。

walterwongbbl, 2024-03-02:
@ChenKe404 非常抱歉。  因为您登录了Bambu账号后，我们默认会使用云服务控制打印机。  如果没有网络，您将只能使用切片功能，但是无法控制打印机或者发送打印。  但是打印机开启Lan模式后，我们将使用局域网控制打印机。您仍然可以发送模型文件进行打印。 目前我们还不能做到自动切换，因为Lan模式需要用户手动输入Pin码连接。

walterwongbbl, 2024-03-02:
我根据您的描述连接了一个没有接入互联网的wifi。尝试操作后没有出现未响应的情况。请问我的操作顺序和您的是否一致。
我的操作步骤是：
1. 打开BambuStudio
2. 登录BambuStudio
3. 断开互联网
4. 切换到设备页面
https://github.com/bambulab/BambuStudio/assets/120699126/8850b4b0-a15e-4729-9aae-7ae8bf7e125e

ChenKe404, 2024-03-02:
> @ChenKe404 非常抱歉。 因为您登录了Bambu账号后，我们默认会使用云服务控制打印机。 如果没有网络，您将只能使用切片功能，但是无法控制打印机或者发送打印。 但是打印机开启Lan模式后，我们将使用局域网控制打印机。您仍然可以发送模型文件进行打印。 目前我们还不能做到自动切换，因为Lan模式需要用户手动输入Pin码连接。

我现在无法测试局域网是否可以使用打印机，因为会出现本issue提到的无响应问题。

ChenKe404, 2024-03-02:
> 我根据您的描述连接了一个没有接入互联网的wifi。尝试操作后没有出现未响应的情况。请问我的操作顺序和您的是否一致。 我的操作步骤是：
> 
> 1. 打开BambuStudio
> 2. 登录BambuStudio
> 3. 断开互联网
> 4. 切换到设备页面
>    https://github.com/bambulab/BambuStudio/assets/120699126/8850b4b0-a15e-4729-9aae-7ae8bf7e125e

请看我录的视频：
1.拔掉路由器的网线，电脑连上wifi。
https://github.com/bambulab/BambuStudio/assets/75658944/c696eb97-4fa7-4b05-9693-84dbdd9df29c
这里我没开打印机，但是开了打印机结果也是一样的，跟打印机无关。

2.完全断网
https://github.com/bambulab/BambuStudio/assets/75658944/127f24f3-e8fb-4852-a8ca-56dc873b965e
这里完全断网，就不会出现无响应情况。

3.在下的拙见:
我觉得问题可能是出在切换到设备标签时，软件会判断网络连接情况，如果网络未连接则不做任何事；而如果网络连接，则向服务器请求服务，但这里的网络请求却是同步请求，如果请求久久不能得到响应而导致UI阻塞，而后响应超时又重新发出请求，最后陷入"死循环"。

ChenKe404, 2024-03-02:
@walterwongbbl 刚才我又尝试了您的步骤，即先打开BambuStudio再断网，这次的情况是：切换到“设备”页，界面开始卡顿，但是这次，卡不了一会儿就会得到"无法连接云服务器"的错误提示，然后界面就恢复到正常状态，不再卡顿。
[image]

walterwongbbl, 2024-03-11:
@ChenKe404 我大致了解了你的需求，目前Studio还不支持登录账号的情况下脱机使用。 我将这个需求加入我们的需求池。
之后和其他开发人员进行讨论后才能确定是否需要支持以及方案。

walterwongbbl, 2024-03-12:
@ChenKe404  这里你也可以不登录我们的Bambu studio，这样就能完全脱机使用打印机了。

closed 2024-05-21 by walterwongbbl