bambulab/BambuStudio #3646: Camera [-90]! error on the X1C

BRadDominator, 2024-02-25:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro version 22H2

### Additional system information

Processor	Intel(R) Core(TM) i9-10850K CPU @ 3.60GHz   3.60 GHz
Installed RAM	64.0 GB (63.9 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab x1c

### How to reproduce

1. Go to device
2. Press play button
3. error [-90]!

### Actual results

[image]

### Expected results

Camera should be able to see print bed.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14397714/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-02-26:
May some as https://github.com/bambulab/BambuStudio/issues/3199
Please reboot the printer and retry.

BRadDominator, 2024-02-26:
When I rebooted there was a firmware update, I ran the update and I can now
view the camera again.  Thank you!