bambulab/BambuStudio #3647: 'too close to others' warning is worse than useless.

AdrianGarside, 2024-02-26:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

N/A

### How to reproduce

Add two objects whose bounding boxes overlap but that are not remotely touching each other.

### Actual results

Incorrect warning that they are too close:
[image]

Because the warning is operating just off bounding boxes it's highly inaccurate. So much so that the warning is worse than useless since a sensible user will be trained to ignore it such that if it were actually ever correct (unlikely since most users wouldn't accidentally overlap objects), the warning may get ignored.

### Expected results

A useful warning when the objects *actually* intersect instead of just their bounding boxes.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-27:
Hi @AdrianGarside 

thanks for reporting this issue

this warning is just used to mention there may has some objects too close, it can be skipped which won't affect the slicing,
it uses the convex hull to do the detection which is not accurate for your nested cases
we had considered to use the real polygons to do the detection before(which will have an accurate result)
however, the performance will be bad

you can just go ahead for slicing and if there are print conflicts, there will be serious warnings
[image]

AdrianGarside, 2024-02-27:
I understand that and I do ignore it. My point was that a warning with such a high false positive rate is worse than useless. I’ve seen a number of posts on Reddit asking about this warning and confusion over it given there was no intersection between the objects. By spamming the user with a warning that is almost always incorrect you make the warning useless in the much rarer case that the user accidentally did have objects that touched unintentionally. Because by that time the chances are high that you’ve successfully trained them to ignore the warning.

TeDsWeb, 2024-02-27:
Instead of the convex hull or real polygons, how about using the alpha shape, which is a generalization of the convex hull and may work better because it can form concave parts and may also be faster than the real polygons. I have no idea if this hasn't already been tested or how well it really works, I just wanted to add this as a suggestion.

lanewei120, 2024-02-27:
thanks for the suggestions

we are also considering remove this warning

eried, 2024-02-29:
I agree this is totally pointless, the automatic arrange should work like https://github.com/Rhoban/Plater and use that logic to warn when things collide

Haidiye00, 2024-03-01:
I have removed this warning, please wait for version V1.9. Thank you

closed 2024-05-06 by Haidiye00