bambulab/BambuStudio #3649: [linux] camera load failed [2]

flying-x, 2024-02-26:
### Bambu Studio Version

bambustudio-git 01.08.02.56.36.gf6930bb6a-1

### Where is the application from?

Built from source by yourself or others

### OS version

manjaro (based on arch linux, rolling)

### Printer

bambu x1c

### How to reproduce

this was working fine for a long time but this week after updating the OS (rolling update from manjaro), this stopped working. I can no longer enable remote monitoring of the printing in the device tab. It shows "load failed [2]", which seems to be related to media player.

you just need to have manjaro (or arch linux) and try this feature there.

### Actual results

video doesn't load

### Expected results

expecting the video to load

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [X] Log file
- [ ] Project file

flying-x, 2024-02-26:
BTW, using the phone app, it works fine.

lanewei120, 2024-02-27:
@flying-x 

1. could you check your gstreamer and plugins firstly, are they still there?
2. how about the flathub image, https://flathub.org/apps/com.bambulab.BambuStudio
3. how about the newest bambustudio release https://github.com/bambulab/BambuStudio/releases/download/v01.08.04.51/BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage or https://github.com/bambulab/BambuStudio/releases/download/v01.08.04.51/BambuStudio_linux_fedora_v01.08.04.51_20240117.AppImage

flying-x, 2024-02-27:
Thanks for your help.

1. How do I check?
2. Flatpak works fine
3. AppImage doesn’t work even building from latest source doesn’t work

lanewei120, 2024-02-27:
we need following gstreamer components
ibgstreamer1.0-dev libgstreamer-plugins-good1.0-dev  libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad

maybe you can use 'sudo pacman -S gstreamer' to do a upgrade?

or check the gstreamer related binary and libraries under
/lib/x86_64-linux-gnu/gstreamer-1.0/

flying-x, 2024-02-27:
Aha, thanks a lot! 

I have the following gst plugin packages installed / not installed:

```
extra/gst-plugins-base-libs 1.22.10-1 (2.2 MiB 11.5 MiB) (Installed)
    Multimedia graph framework - base
extra/gst-plugin-libcamera 0.2.0-1 (36.2 KiB 94.8 KiB)
    Multimedia graph framework - libcamera plugin
extra/gst-plugins-bad-libs 1.22.10-1 (2.4 MiB 10.3 MiB) (Installed)
    Multimedia graph framework - bad
extra/gst-plugin-pipewire 1:1.0.3-1 (48.2 KiB 123.2 KiB) (Installed)
    Multimedia graph framework - pipewire plugin
extra/gst-plugins-espeak 0.5.0-4 (12.7 KiB 33.9 KiB)
    Use espeak as a sound source for gstreamer
extra/gst-plugin-opencv 1.22.10-1 (88.6 KiB 245.6 KiB)
    Multimedia graph framework - opencv plugin
extra/gst-plugins-good 1.22.10-1 (2.2 MiB 7.0 MiB) (Installed)
    Multimedia graph framework - good plugins
extra/gst-plugins-ugly 1.22.10-1 (210.4 KiB 654.4 KiB) (Installed)
    Multimedia graph framework - ugly plugins
extra/gst-plugin-qmlgl 1.22.10-1 (58.0 KiB 166.3 KiB)
    Multimedia graph framework - qmlgl plugin
extra/gst-plugins-base 1.22.10-1 (320.2 KiB 967.7 KiB) (Installed)
    Multimedia graph framework - base plugins
extra/gst-plugin-msdk 1.22.10-1 (86.6 KiB 231.6 KiB)
    Multimedia graph framework - msdk plugin
extra/gst-plugins-bad 1.22.10-1 (1.1 MiB 3.6 MiB) (Installed)
    Multimedia graph framework - bad plugins
extra/gst-plugin-qml6 1.22.10-1 (34.0 KiB 86.8 KiB)
    Multimedia graph framework - qml6 plugin
extra/gst-plugin-gtk 1.22.10-1 (36.1 KiB 112.1 KiB) (Installed)
    Multimedia graph framework - gtk plugin
extra/gst-plugin-qsv 1.22.10-1 (118.8 KiB 318.4 KiB)
    Multimedia graph framework - qsv plugin
extra/gst-plugin-wpe 1.22.10-1 (33.8 KiB 92.2 KiB)
    Multimedia graph framework - wpe plugin
extra/gst-plugin-va 1.22.10-1 (159.6 KiB 397.9 KiB)
    Multimedia graph framework - va plugin
```

Do you know which one is needed that I don't already have it installed?

From this page: https://wiki.archlinux.org/title/GStreamer, it seems that I have the good/bad/base already installed.

Also, these AUR packages don't have any gst plugin packages as dependencies. Once I figured this out, I can flag those packages and let them know which plugin package is needed.

https://aur.archlinux.org/packages/bambustudio-git
https://aur.archlinux.org/packages/bambustudio

flying-x, 2024-02-27:
I have a feeling that it is the `msdk` plugin. I will give it and others a try to see if that fixes the problem or not.

flying-x, 2024-02-27:
This is very strange. I can't reproduce this problem anymore. It is my bad. closing.

closed 2024-02-27 by flying-x