bambulab/BambuStudio #3652: Latest Bambu Network Plugin crashes OrcaSlicer on Linux

WayneManion, 2024-02-27:
### Bambu Studio Version

999

### Where is the application from?

Built from source by yourself or others

### OS version

Arch Linux AMD64

### Additional system information

Multiple computers, all running Linux with lots of memory. 

### Printer

P1P converted to P1S

### How to reproduce

Running Linux
Install OrcaSlicer. It works fine.
Install latest Bambu Network Plugin (BNP) through OrcaSlicer. OrcaSlicer still runs fine.
Login to Bambu account via BNP. OrcaSlicer segfaults.
Re-open OrcaSlicer. OrcaSlicer segfaults every time. 

### Actual results

OrcaSlicer segfaults when logged into Bambu account with latest Bambu Network Plugin. 

### Expected results

OrcaSlicer should not segfault when logged into Bambu account. OrcaSlicer worked fine with older versions of Bambu Network Plugin. 

### Project file & Debug log uploads

cdsv

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-27:
how about the result when running bambu studio,
for example https://github.com/bambulab/BambuStudio/releases/download/v01.08.04.51/BambuStudio_linux_ubuntu_v01.08.04.51_20240117.AppImage

WayneManion, 2024-02-27:
The latest Bambu Network Plugin (BNP) seems to work OK in the latest version of BambuStudio. 

The latest version BNP always crashes OrcaSlicer. The old version BNP was fine in OrcaSlicer. 

I created the issue here because I could not find a Github page for BNP itself.

lanewei120, 2024-02-28:
please upload a log file before the crash happens 
thanks