bambulab/BambuStudio #3656: "Fix Model" removes model colors

ikeroy, 2024-02-28:
### Bambu Studio Version

1.8.5.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

1. Color a model with broken geometry/non-manifold edges using the color painting tool
2. Right click the model and click "Fix Model"

### Actual results

The colors on the model after fixing are removed

### Expected results

The model is fixed and colors stay

### Project file & Debug log uploads

[zeraora_color_test_2.zip](https://github.com/bambulab/BambuStudio/files/14437886/zeraora_color_test_2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-02-29:
@ikeroy 

thanks for reporting this issue
currently the fix model will regenerate the triangles outside, and reload it into bambustudio

as the triangles have been changed
the color data can not be used anymore

we are hard to keep the colors after repair

suggest to do the repair before painting
and we will investigate more to check whether we can keep the color data

closed 2024-05-09 by Haidiye00

JoshuaGessner, 2025-08-04:
@lanewei120 This issue with painting after repair, is that this happens with models created by the Image to Model applet on MakerWorld. So they auto generate colors for us and they aren't even usable models at that point.