bambulab/BambuStudio #3657: Feature Request: Layer Speed Normalisation Tower

kvnper, 2024-02-29:
A slicer option to add an automatically generated **tower** with varying **density** and/or **print speed** which normalises the layer time so that all layers are printed at the same PRINT SPEED and with the same/similar LAYER TIME. This means that layers do not need to slow down, causing: 
(1) Inconsistent surface finish (shiny vs matte), and 
(2) Overheating from the nozzle when using difficult materials (occurs with some brands of ABS, PA, etc).

It would be a tower that is automatically generated by the slicer, where the DENSITY and the PRINT SPEED of the tower at each layer varies in order to normalise the model's PRINT SPEED and LAYER TIME.

_Why is this better than/What is the benefit over turning off the setting 'slow down printing for better layer cooling'?_ 
I print a lot of things for other people in all sorts of material, and a good amount of models that are also not designed for FDM 3D printing. What I've found is that the best way, for me, to prevent overheating issues and an inconsistent layer surface finishes from the speed slow down, is to add a cube the same height as the model in order to make the short layers longer. The downside to this is that all other layers also get this additional time, which adds more print time and material than necessary.

_Why is this better than increasing the fan speeds?_
There are a few reasons:
1. Sometimes you cannot increase the fan speed any more, it's already at maximum.
2. Sometimes the right side of the model is worse because of the aux fan only being on the left. Not to mention that some printers don't have an aux fan.
3. Some materials do not play well with higher fan speeds.
4. The issue of an inconsistent surface finish still remains, which is caused by short layers being printed at a different speed.

Rough drawing of the tower concept, diagonal lines represent the increase/decrease in density/speed:
[image]

wrightsh, 2024-07-25:
This would be useful for more than just tall parts.  Here's an example of a short rectangular part that has some holes going through it.  The part on the right was printed as a single unit, and has very strong banding due to differences in layer time.  The one on the left looks much better, and has flatter sidewalls because I printed two at once so that the faster layers had more time to cool.  I was thinking it might work well to set a minimum time per layer, and slow down the speed of the infill to meet that time, that way you don't need the extra tower and going slow with the infill shouldn't make any difference.

[image]

DrFischie, 2024-10-06:
Yes, this is something we need. Please add something like this.

Fitz0uille, 2024-12-03:
Definitely something that can help to have a constant look on you print.