bambulab/BambuStudio #3658: Flatpack version having error on loading some stls

JinicMedia, 2024-02-29:
### Bambu Studio Version

BambuStudio_linux_ubuntu_v01.08.04.51_20240117

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 21

### Additional system information

System:
  Kernel: 6.5.0-21-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 6.0.4 tk: GTK 3.24.33
    wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop Mobo: ASUSTeK model: TUF Z390-PLUS GAMING (WI-FI) v: Rev X.0x
    serial: <superuser required> UEFI: American Megatrends v: 2808 date: 05/27/2020
CPU:
  Info: 6-core model: Intel Core i7-8700 bits: 64 type: MT MCP smt: enabled arch: Coffee Lake
    rev: A cache: L1: 384 KiB L2: 1.5 MiB L3: 12 MiB
  Speed (MHz): avg: 1200 min/max: 800/4600 cores: 1: 1200 2: 1200 3: 1200 4: 1200 5: 1200
    6: 1200 7: 1200 8: 1200 9: 1200 10: 1200 11: 1200 12: 1200 bogomips: 76799
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: NVIDIA GP106 [GeForce GTX 1060 3GB] vendor: Micro-Star MSI driver: nvidia
    v: 535.161.07 pcie: speed: 8 GT/s lanes: 16 ports: active: none off: HDMI-A-1,HDMI-A-2
    empty: DP-1,DP-2,DVI-D-1 bus-ID: 01:00.0 chip-ID: 10de:1c02 class-ID: 0300
  Device-2: Logitech HD Pro Webcam C920 type: USB driver: snd-usb-audio,uvcvideo bus-ID: 1-11:5
    chip-ID: 046d:082d class-ID: 0102 serial: <filter>
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia display-ID: :0 screens: 1
  Screen-1: 0 s-res: 3840x1080 s-dpi: 93 s-size: 1049x292mm (41.3x11.5") s-diag: 1089mm (42.9")
  Monitor-1: HDMI-0 pos: primary,left res: 1920x1080 hz: 60 dpi: 94 size: 521x293mm (20.5x11.5")
    diag: 598mm (23.5")
  Monitor-2: HDMI-1 pos: right res: 1920x1080 hz: 60 dpi: 94 size: 521x293mm (20.5x11.5")
    diag: 598mm (23.5")
  OpenGL: renderer: NVIDIA GeForce GTX 1060 3GB/PCIe/SSE2 v: 4.6.0 NVIDIA 535.161.07
    direct render: Yes
Audio:
  Device-1: Intel Cannon Lake PCH cAVS vendor: ASUSTeK driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:a348 class-ID: 0403
  Device-2: NVIDIA GP106 High Definition Audio vendor: Micro-Star MSI driver: snd_hda_intel
    v: kernel pcie: speed: 8 GT/s lanes: 16 bus-ID: 01:00.1 chip-ID: 10de:10f1 class-ID: 0403
  Device-3: Logitech HD Pro Webcam C920 type: USB driver: snd-usb-audio,uvcvideo bus-ID: 1-11:5
    chip-ID: 046d:082d class-ID: 0102 serial: <filter>
  Sound Server-1: ALSA v: k6.5.0-21-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Cannon Lake PCH CNVi WiFi driver: iwlwifi v: kernel bus-ID: 00:14.3
    chip-ID: 8086:a370 class-ID: 0280
  IF: wlo1 state: down mac: <filter>
  Device-2: Intel Ethernet I219-V vendor: ASUSTeK driver: e1000e v: kernel port: N/A
    bus-ID: 00:1f.6 chip-ID: 8086:15bc class-ID: 0200
  IF: eno2 state: up speed: 1000 Mbps duplex: full mac: <filter>
  IF-ID-1: br-06c219106e6a state: down mac: <filter>
  IF-ID-2: br-1ef4a90f544b state: down mac: <filter>
  IF-ID-3: br-2b6ab5038b51 state: down mac: <filter>
  IF-ID-4: br-462068b3f76d state: up speed: 10000 Mbps duplex: unknown mac: <filter>
  IF-ID-5: br-597f5495116e state: down mac: <filter>
  IF-ID-6: br-6def2651c122 state: down mac: <filter>
  IF-ID-7: br-a17824f5c3f9 state: down mac: <filter>
  IF-ID-8: br-aef341b07b5f state: down mac: <filter>
  IF-ID-9: br-b4e814fce1e1 state: up speed: 10000 Mbps duplex: unknown mac: <filter>
  IF-ID-10: docker0 state: down mac: <filter>
  IF-ID-11: veth54ac199 state: up speed: 10000 Mbps duplex: full mac: <filter>
  IF-ID-12: veth6ca2631 state: up speed: 10000 Mbps duplex: full mac: <filter>
[... 96 lines cut ...]
Bambulab X1C

### How to reproduce

Load an stl,

### Actual results

Message Loading of a model file failed.
The file does not contain any geometry data.

Do the same with the AppImage same version, no errors.

### Expected results

The file is loaded.

### Project file & Debug log uploads

[debug_Thu_Feb_29_10_44_37_4.log.0.zip](https://github.com/bambulab/BambuStudio/files/14446058/debug_Thu_Feb_29_10_44_37_4.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-02-29:
@JinicMedia 
could you share the stl link here?
thanks

JinicMedia, 2024-02-29:
After upload and downloading them on a serveur and opening again, I saw the problem was'nt the file, but were I've putted it.

If it's on my local hard drive, it's ok, but does'nt work on a shared drive from my nas in smb protocol. 

But, really work fine with the app image.

lanewei120, 2024-06-26:
need to report a issue on https://github.com/flathub/com.bambulab.BambuStudio
close it here