bambulab/BambuStudio #3660: Camera not working on my pc

ellisdallas1985, 2024-02-29:
### Bambu Studio Version

1.8

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu x1 carbon

### How to reproduce

click on device tab and press play for camera

### Actual results

failed -90 

### Expected results

camera to show up 

### Project file & Debug log uploads

idk what this is

### Checklist of files to include

- [X] Log file
- [ ] Project file

ellisdallas1985, 2024-02-29:
I have tried to delete and reload bambu studio as well as restarting my pc and nothing has helped

bambu123, 2024-03-01:
May some as https://github.com/bambulab/BambuStudio/issues/3199
Please reboot the printer and retry.