bambulab/BambuStudio #3661: Export STLs

ZiggyHentze, 2024-03-01:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu P1P

### How to reproduce

1. Cut model into 12 parts with dowels
2. Export each as STL parts
3. Open in other software (ZBrush in my case, 3D viewer and MeshMixer) and the holes for Dowels are missing?

### Actual results

Holes just gone?

### Expected results

I need to print over multiple printers as the job is to huge - "Print Farm" via friends with printers

### Project file & Debug log uploads

Should be clear than to still send the models too

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ZiggyHentze, 2024-03-01:
The dowels export but the parts after exporting to STL and then re importing into Studio are all missing same as other programs?
[image]

Haidiye00, 2024-03-06:
[image]
This issue has been fixed, please wait for version 1.9.Thank you for your discovery and support

closed 2024-05-06 by Haidiye00