bambulab/BambuStudio #3662: When uploading print profile to makerworld, I get "did not pass the slicing test."

jetpad, 2024-03-01:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.1.2 (23B92)

### Additional system information

M1 Macbook

### Printer

Bambu Lab X1C

### How to reproduce

I slice and print a model on the X1C and then upload the 3MF file to makerworld.com and it says 

Your print profile "Multicolor Fidget Spinner" for "The Vortex Kit Card" did not pass the slicing test. Please make sure the model is sliced normally on Bambu Studio before uploading.

### Actual results

I'm not able to public the print profile because it fails a slicing test on makerworld. 

### Expected results

It should accept the print profile. 

### Project file & Debug log uploads

[debug_Fri_Mar_01_08_16_40_7633.log.0.zip](https://github.com/bamb
[TheVortexV31.3mf.zip](https://github.com/bambulab/BambuStudio/files/14462206/TheVortexV31.3mf.zip)
ulab/BambuStudio/files/14462205/debug_Fri_Mar_01_08_16_40_7633.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2024-03-05:
hello @jetpad 
is the attached 3mf the same with the one you uploaded to makerworld?

I just tried it to upload as a private test, it can upload successfully

could you give a snapshot of your failure

jetpad, 2024-03-05:
Yes, the one that was attached is the same one that I sent you earlier with
the video snapshot.

lanewei120, 2024-05-30:
can upload successfully now

closed 2024-05-30 by lanewei120