bambulab/BambuStudio #3663: Initial layer height setting has no effect when printing on the raft

kmeldner, 2024-03-01:
### Bambu Studio Version

1.8.4.5.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1 Carbon

### How to reproduce

1) Set initial layer height to different value than layer height
2) activate raft 

### Actual results

Initial layer height setting has no effect when raft is active. 
This setting only works on build plate directly.

### Expected results

initial layer height setting should work on build plate and on raft

### Project file & Debug log uploads

[plate 04 base_C.zip](https://github.com/bambulab/BambuStudio/files/14465570/plate.04.base_C.zip)
[debug_network_Fri_Mar_01_17_53_05.log.zip](https://github.com/bambulab/BambuStudio/files/14465573/debug_network_Fri_Mar_01_17_53_05.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-03-07:
[image]
Could you take a screenshot to show the symptoms of your problem? I can't reproduce this problem.

kmeldner, 2024-03-12:
Ok, I see there is a slight misunderstanding.
Let me than explain the problem.

When printing with thin layers and needing to use a raft there are problems with quality of the bottom of the part.
This is because thin layers can not support themselves between raft lines.
solution to this is not making 0 clearance between raft and the object as in such case raft is impossible to remove.

therefore having bottom most layer of the part printed in thicker layer would allow to create support for subsequent thin layers and have ability to remove raft.

I thought that initial layer setting concerns the model and not the raft. But apparently it does not.

Another problem is raft layer height when setting global layer height to some small value. in such case it is still desirable to print raft with thick layers to quickly get raft height needed.

So I actually see following problems here:
1) having setting available for each object that would allow to define bottom most layer of this object for slicing.
2) having initial layer setting, that is already there, that allows to define first layer on the build plate thickness
3) having setting for layer height of the raft.

This is technically not a bug at this time, but definitely would make life easier. 
Currently I am struggling a lot and have to use various workarounds to achieve something that should be easy.

Cris

zhimin-zeng-bambulab, 2024-03-12:
Thanks for your explanation!