bambulab/BambuStudio #3664: 希望加入鼠标对模型框选功能

ChenKe404, 2024-03-02:
**Is your feature request related to a problem? Please describe.**
我有时需要打印许多小零件，软件的自动整理结果往往不能满足我的需要，当我需要整体移动多个零件时，总是需要按住ctrl键进行复选，但是如果一不小心没选到，那么之前选定的零件也会被取消选定，这令我感到十分不适。又因经常使用建模软件，而大部分建模软件左键按下拖拽都默认是框选功能，而BambuStudio却是旋转视角，这同样也令我头疼。

**Describe the solution you'd like**
我希望软件能增加模型的框选功能，让我在场景中拖拽左键能显示一个选框，将被框住的所有零件设为选中状态。就像多数建模软件那样。

**Describe alternatives you've considered**
最好将左键的拖拽设为框选，中键设为旋转视角，右键设为移动视角。另外，目前软件的中键和右键拖拽都是移动视角，这就浪费了一个按键的功能性，或者预设几套鼠标动作供用户自行选择。

**Additional context**
[image]
[效果如以上PS图所示]
以上，感谢各位

track-tech, 2025-03-28:
同希望这样

track-tech, 2025-03-28:
按住shift可以这样