bambulab/BambuStudio #3671: Filament PA Profile for AMS not persisting after selection

GWren70, 2024-03-03:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.2.1

### Printer

Bambu Lab X1C w/ AMS

### How to reproduce

The issue is a bit random at times, but these are the two methods that I have consistently been able to reproduce it with.

On Printer:
1. Go to Filamants
2. Select the desired filament and edit it
3. Select a previously created PA Profile and Confirm
4. Select another filament
5. Select the first filament again and edit and the selected value is reset to whatever profile matches the name of the filament

In Bambu Studio
1. Select desired device
2. On the Status screen select the filament and edit it
3. Select a previously created PA Profile and Confirm
4. Leave the Device tab
5. Go back to the device tab
6. The PS profile previously selected will reset to whatever matches the filament name
7. Alternately, sometimes you can just watch it change if you stay on the device tab

I have 3 combos, and this happens on all of them.  Since I must use Generic PLA on brands that you have not included, having multiple PA Profiles is great convenience, but only if I can load the filament, select it, and walk away knowing it is good.  Sometimes it appears to stick, but will still drop at some point.  It will stick if I calibrate it in, and then not touch it again, but that defeats the purpose of being able to store multiple...just force us to calibrate it in and like you used to.

### Actual results

The PA Profile selected does not persist as it should.

### Expected results

The PA Profile I select when loading filament should persist until I load new filament and change the value, or calibrate a new filament in that slot.

### Project file & Debug log uploads

No file required.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GWren70, 2024-03-03:
As a side note, it would be nice if you also gave us a way to edit that list of stored PA Profiles to allow cleanup of no longer needed or duplicate results.  Since these are tied to the filament selected, it would also be nice if I could create a filament profile on the device so that I do not need to use a Generic PLA and rely on a long list of saved profiles.

zhimin-zeng-bambulab, 2024-03-08:
Thank you for reporting this issue, we will improve it in the future.

0verEngineer, 2025-08-28:
@BambulabRobot Here is your f**king response

0verEngineer, 2025-08-28:
@zhimin-zeng-bambulab Can you say something about this issue? Why is it not fixed - not even touched once since a year?!

0verEngineer, 2026-04-08:
> Based on your description, the settings are also being reset to default values on the printer side, which suggests this is most likely a firmware issue.

To me it seems more like it COULD be like this:
1. We set the PA Profile in Bambu Studio
2. Bambu Studio fails to transfer it to the Printer 
3. It is reset in Bambu Studio

I had this issue previously in Orcaslicer as well, so it could be in the Code that Bambu Studio and Orcaslicer still have in common, or (my take) it is in the Network Plugin.