bambulab/BambuStudio #3672: Next button disappears when doing Flow Dynamics and Flow Rate calibrations.

IKnowImLost224, 2024-03-03:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Do the Flow Rate or Flow Dynamics Calibration. Once the print is completed the "Next" button is not available to go to the next page and enter the results.

### Actual results

[image]

### Expected results

The "Next" button should appear to go to the next step. The only way I can get it to show up is to start a new print, then cancel it from the printer.
[image]
 

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14474365/log.zip)
[Flow Calibration.zip](https://github.com/bambulab/BambuStudio/files/14474366/Flow.Calibration.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-03-05:
@IKnowImLost224  If you minimize the software first and then open it, will this button appear?

IKnowImLost224, 2024-03-05:
I will try this with the next filament I need to tune, but I don’t believe so. I know I tried a few different things and the only way it seemed to work was to re-send the calibration to the printer then cancel it.

Orochium, 2024-03-06:
I have this same issue now, and have had to stop most printing as I cannot add and calibrate the new filament I picked up

P1P + P1S Official Upgrade Kit, Firmware 01.05.02.00
Bambu Slicer Version 1.8.4.51

No Next button or Printer doesn't report when it has completed printing of calibration G-Code, resulting in the inability to complete the process. 

[image]

zhimin-zeng-bambulab, 2024-03-07:
Same as https://github.com/bambulab/BambuStudio/issues/2364

We will investigate the cause of this problem, until then, you could manually calculate it yourself.

new flow rate = (current flow rate) * (1 + ((number above smoothest face)/100))

So if your current flow rate is .98, and my smoothest face is 10, it would be:

.98*(1+(10/100)) = 1.078

IKnowImLost224, 2024-03-07:
I actually just did a calibration on a new filament today and it seemed to work fine. Not sure if it’s an intermittent problem?

Orochium, 2024-03-08:
Tried this today and it appears to all be working suddenly, no versions changed

Guessing something on the cloud printing side was catching it somehow? Can only speculate I suppose; appreciate the hard work all the same. Appears to work with every profile I've tested now (both Bambu and non-bambu filaments)

IKnowImLost224, 2024-03-09:
I tried both tests again this morning and ran into the same issue again. Next button disappeared on both tests.

Orochium, 2024-03-09:
Yep, just tried to calibrate a new roll of filament and it's gone today for me again as well

Orochium, 2024-03-09:
I had to completely quit the program, and reopen it immediately to the calibration tab I was on before, and in doing so, it reopened to the same page and the next button reappeared.

IKnowImLost224, 2024-03-09:
Yeah I figured out my issue after I replied unfortunately. I tried to do the calibration from an old BLS I had open. After I closed that and re-opened a new instance of BLS I didn’t have any issues.

Orochium, 2024-03-09:
Yeah, this is weird, on mine, doesn't matter, I leave the program open a lot since I'm constantly using it, I shouldn't have to close all the way out of the program, and reopen a new instance, just to get the "Next" button to appear, so somewhere or another, it's not refreshing the screen consistently after the calibration print finishes

IKnowImLost224, 2024-03-09:
Yeah I get what you’re saying. I’m just thinking if there was an update, maybe it came after I had that instance of BLS open. After I re-opened it I didn’t have a problem

LiZ-Li-BBL, 2024-05-15:
We've fixed this issue in the latest release. https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.66.

closed 2024-09-14 by LiZ-Li-BBL