bambulab/BambuStudio #3674: Very small text on flow calibration

SteveAtSmartersoft, 2024-03-04:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Printer

X1c

### How to reproduce

Go to Calibration
Run it
See the text on success

### Actual results

The text is very tiny quite often. (as per the screenshot)  But not always.
[image]

### Expected results

Larger font

### Project file & Debug log uploads

No project

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-03-13:
Thanks for reporting this issue, we will also try to reproduce this issue.

SteveAtSmartersoft, 2024-03-13:
It does not happen every time, but I've seen it 3-4 times now.  Like every 3rd time or so.

Best regards
___________________________________________
Steve Horvath
Principal Consultant

SmarterSoft
Level 17, Angel Place,
123 Pitt St Sydney NSW 2000

e: ***@***.******@***.***>
w: www.smartersoft.com.au<http://www.smartersoft.com.au/>

SmarterSoft is a registered trading name of Areanet Pty. Ltd.

We acknowledge the Traditional Owners and Custodians of the lands on which we work and pay our respects to Indigenous Elders past, present and emerging.

LiZ-Li-BBL, 2024-05-15:
Hi,
please use the latest release, we've fixed this issue. https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.66.

closed 2024-05-15 by LiZ-Li-BBL