bambulab/BambuStudio #3677: MacOS Text not showing correctly

WNTLtd, 2024-03-04:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Monterey V12.7.2

### Printer

Bambu Lab X1C

### How to reproduce

Generating Text on the surface of my model - and some letters are backwards and or vertical!

### Actual results

[image]

### Expected results

The text should be written as entered.

### Project file & Debug log uploads

[logs.zip](https://github.com/bambulab/BambuStudio/files/14484739/logs.zip)
[Export for BLabs.3mf.zip](https://github.com/bambulab/BambuStudio/files/14484741/Export.for.BLabs.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-03-12:
Thanks for reporting this issue, we will investigate the cause of this problem.

Haidiye00, 2024-08-01:
Hello, can I record a video or GIF? Thank you

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00