bambulab/BambuStudio #3679: Elephant Foot Compensation A1 Mini has been removed after 1.7.7.89

JonLaRue, 2024-03-04:
### Bambu Studio Version

> 1.7.7.89

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10 + Linux

### Printer

Bambu Lab A1 Mini

### How to reproduce

Look at print Profiles for A1 Mini in Bambu Studio, all no longer have Elephant's foot comp. I am having really bad elephant's foot on my print since upgrading. 

[image]
[image]

### Actual results

Current Release of Bambu Studio has "0" for Elephant's Foot Compensation.

### Expected results

Bambu Studio Release 1.7.7.89 has "0.15" maintained for Elephant's Foot Compensation. This is what should be in the Slicer.

### Project file & Debug log uploads

[3MF for Elephant Foot.zip](https://github.com/bambulab/BambuStudio/files/14486793/3MF.for.Elephant.Foot.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

BillyBigStuff, 2024-10-20:
Has this been fixed yet?

MattW42, 2025-03-22:
Why hasn't this issue been resolved?
Getting elephant foot issues on A1 mini.
Seems like a simple fix to update the default slicer profiles.