bambulab/BambuStudio #3681: Color Bleed in slicer

Tedwell, 2024-03-06:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 10

### Additional system information

Dell Alienware Area 51m R1  CPU: Intel core i9-9900k   Memory: 32gb  GPU: Nvidia 2080 

### Printer

X1 Carbon

### How to reproduce

Slice plate of color lettered model.

### Actual results

Layer 48 out of 50 has color errors outside of colored letters. Like a bleed in sliced lines. Occurs in printed model as well.  Seems to happen with coloring of neighboring characters and not where the bleed is. So far it has only happened to this particular model. Other prints have come out clean.
[X1C Color Error in slicer.docx](https://github.com/bambulab/BambuStudio/files/14513414/X1C.Color.Error.in.slicer.docx)

### Expected results

clean colored letters

### Project file & Debug log uploads

[Test tube holder names v9.zip](https://github.com/bambulab/BambuStudio/files/14513718/Test.tube.holder.names.v9.zip)
[debug_Mon_Mar_04_10_09_45_19804.log.zip](https://github.com/bambulab/BambuStudio/files/14513721/debug_Mon_Mar_04_10_09_45_19804.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Tedwell, 2024-03-06:
Nvidia GeForce RTX 2080

Tedwell, 2024-03-06:
I increased the size of the model to 100.01% with no bleed. Yay!!! Now off to print to see if it prints correctly as well. Still want to know why it bled in the first place. Tolerances are crucial to this model.

Tedwell, 2024-03-06:
It printed fine. No bleed. I redesigned characters in Fusion 360. Loaded it into Bambu Studio. Colored the characters in Bambu Studio with fill. Same as before. Still bleeds. Increased scale to 100.01% and no bleed. Did I just find a glitch or is something else going on.

Thanks

Tedwell, 2024-03-07:
[image]
Top pic at layer 48/50. Middle pic at layer 50/50. Bottom pic is a good print by increasing scale to 100.01%.

Tedwell, 2024-03-10:
Anyone here to comment on this issue?

zhimin-zeng-bambulab, 2024-03-11:
Thanks for reporting this issue, it is a BUG and we are investigating the cause.

MonsterMaxx, 2024-03-12:
I have something that's tan w/ red lettering.  Can clearly see the bleed.
[image]

Tedwell, 2024-03-13:
> I have something that's tan w/ red lettering. Can clearly see the bleed. I have not yet printed this again so I don't know what the solution would be. ![PXL_20240304_163324575](https://private-user-images.githubusercontent.com/88036569/312196161-2e863466-f60a-4673-9969-d6e2d6b01902.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyOTE2MjksIm5iZiI6MTcxMDI5MTMyOSwicGF0aCI6Ii84ODAzNjU2OS8zMTIxOTYxNjEtMmU4NjM0NjYtZjYwYS00NjczLTk5NjktZDZlMmQ2YjAxOTAyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEzVDAwNTUyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3NDNlZGM5MDUzMTZiOWRiZDg4MDZhYjRjN2EwMTI3MzBlYTAzOTQ5OTc0NWY0YTQ4NmZjMzE1ZjJjODU4YzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZXp53WDz__7wCLGURGQ2YahI-mdY89WlMlRGlgrHYjs)

Does it look the same in Bambu Studio?

zhimin-zeng-bambulab, 2024-03-13:
@MonsterMaxx  Could you share your 3mf file?

MonsterMaxx, 2024-03-13:
> @MonsterMaxx Could you share your 3mf file?

no I can't share, sorry, the stls are confidential.

MonsterMaxx, 2024-03-13:
[image]
in the slicer it looks clean, no bleed, but as you can see by the actual part it's pretty bad.

Can I get control over 'retraction'??  I think a little more retraction might solve the problem.

zhimin-zeng-bambulab, 2024-03-13:
[image]
You can try to increase this value.

Tedwell, 2024-03-13:
> Thanks for reporting this issue, it is a BUG and we are investigating the cause.

Thank you for this information. Please let us know the outcome.

zhimin-zeng-bambulab, 2024-03-14:
> Thank you for this information. Please let us know the outcome.
we will fix it in the next version

EricDimitri, 2024-03-17:
> > Thank you for this information. Please let us know the outcome.
> 
> we will fix it in the next version

When will next version be released? Is there an ETA?

Thank you!!

zhimin-zeng-bambulab, 2024-03-27:
it has been fixed in v 01.09.00.06

closed 2024-03-27 by zhimin-zeng-bambulab

Tedwell, 2024-03-31:
> I have something that's tan w/ red lettering. Can clearly see the bleed. I have not yet printed this again so I don't know what the solution would be. [...]

Could this be caused by "Reduce Waste during Filament Change" setting? https://wiki.bambulab.com/en/software/bambu-studio/reduce-wasting-during-filament-change

zhimin-zeng-bambulab, 2024-04-02:
> Could this be caused by "Reduce Waste during Filament Change" setting? https://wiki.bambulab.com/en/software/bambu-studio/reduce-wasting-during-filament-change

Maybe, since there is no 3mf file, I don't know how your parameter settings. 
You could try to increase the flushing volume, or don't choose "flush into object's infill".