bambulab/BambuStudio #3682: Add scarf joint to minimize seam visibility

TeDsWeb, 2024-03-06:
OrcaSlicer has just added a new seam connection to minimize the visibility of the seam. The results so far look quite impressive and I think it would also be an improvement for BambuStudio.

This [issue](https://github.com/SoftFever/OrcaSlicer/issues/3211) describes the new seams in more detail, and this [pull request](https://github.com/SoftFever/OrcaSlicer/pull/3839), which was recently merged, also provides more information about the new seam connection.

SaltWei, 2024-03-18:
@TeDsWeb Thank you very much. We are doing this, and will be published with next version soon.

QingZhangBambu, 2024-04-01:
@TeDsWeb This function has been added to release version 1.9 beta 1, you can download and try it.