bambulab/BambuStudio #3686: Wrong AMS slot chosen after slicing

EhRabz, 2024-03-08:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

P1s

### How to reproduce

Load AMS slot 1 and 2 with black rolls, set colour 1 as the chosen colour before slice. After slice it defaults to A2 and has to be switched back to A1 before hitting print.

### Actual results

Tries to print from A2 before A1 , even if they're the exact same filament / colour 

### Expected results

It uses the first slot first..

### Project file & Debug log uploads

sd

### Checklist of files to include

- [ ] Log file
- [ ] Project file

EhRabz, 2024-03-08:
My last thread was closed and marked as " solved " , but that's not the case lol.

QingZhangBambu, 2024-03-11:
could you provide 3mf files?

bam-winterfell, 2024-03-19:
hi @EhRabz 
I replied to you on # 3285, you can see if it solves your problem.
[https://github.com/bambulab/BambuStudio/issues/3285](url)

closed 2024-10-17 by walterwongbbl