bambulab/BambuStudio #3692: Timelapse Video Format & File Type

CaptBeard, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
When I download the timelapse videos captured during a print, it's in .avi format and I have to convert it to play it natively on my Mac.

**Describe the solution you'd like**
A great solution would be the choice for more supported formats of video files. Like .mov, .mp4. or even .gif.

**Describe alternatives you've considered**
So far I've been converting all the timelapse videos to h265 .mov files. This is an added step that I wish didn't exist.

**Additional context**
This is "issue" is only present when using BambuStudio on a macOS computer.

dbshort, 2024-05-28:
Same. I also cannot play the .avi files on my Mac without re-encoding them using a 3rd party video player. Please change the timelapse video format to h264 or h265 with .mov or .mp4 format for universal support on Mac and PC.

petetum, 2026-09-02:
For mac users I have written a script. This does not re-encode the video files but fixes them without quality loss.
Just download it, put it in the same folder where the avi files are and double click on it.
It will scream on you saying it cannot verify it.
Open System Settings, Privacy and security, scroll down all the way and click on Open Anyway button.

https://drive.google.com/file/d/1nWGO1AqlqS4j5Qx3hhXK1zb1BFbGifN8