bambulab/BambuStudio #3694: CMake error. OpenVDB could not be found with the bundled find module.

Christian-Kauth, 2024-03-09:
### Bambu Studio Version

master

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 22.04.4 LTS

### Printer

Bambu Lab X1

### How to reproduce

```
cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/chris/work/projects/BambuStudio_dep/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release
cmake --build . --target install --config Release -j
```
causes an errror. 

### Actual results

```
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 3
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: /home/chris/work/projects/BambuStudio_dep/usr/local (from cache or command line)
-- PROJECT_SOURCE_DIR: /home/chris/work/projects/BambuStudio
-- CMAKE_MODULE_PATH: /home/chris/work/projects/BambuStudio/cmake/modules/
-- LIBDIR: /home/chris/work/projects/BambuStudio/src
-- LIBDIR_BIN: /home/chris/work/projects/BambuStudio/build/src
-- Found Boost: /home/chris/work/projects/BambuStudio_dep/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams 
-- Boost::boost exists
-- TBB found in /home/chris/work/projects/BambuStudio_dep/usr/local/lib/cmake/TBB
OpenSSL include dir: /home/chris/work/projects/BambuStudio_dep/usr/local/include
OpenSSL libraries: /home/chris/work/projects/BambuStudio_dep/usr/local/lib64/libssl.a;/home/chris/work/projects/BambuStudio_dep/usr/local/lib64/libcrypto.a;dl
-- FindGLEW: did not find GLEW CMake config file. Searching for libraries.
-- FindGLEW: GLEW_USE_STATIC_LIBS: ON.
-- FindGLEW: GLEW_INCLUDE_DIR: /home/chris/work/projects/BambuStudio_dep/usr/local/include
-- FindGLEW: GLEW_INCLUDE_DIRS: /home/chris/work/projects/BambuStudio_dep/usr/local/include
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for SHARED: .so
-- FindGLEW: CMAKE_FIND_LIBRARY_SUFFIXES for STATIC: .a
-- FindGLEW: GLEW_SHARED_LIBRARY_RELEASE: GLEW_SHARED_LIBRARY_RELEASE-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_RELEASE: /home/chris/work/projects/BambuStudio_dep/usr/local/lib/libGLEW.a
-- FindGLEW: GLEW_SHARED_LIBRARY_DEBUG: GLEW_SHARED_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY_DEBUG: GLEW_STATIC_LIBRARY_DEBUG-NOTFOUND
-- FindGLEW: GLEW_SHARED_LIBRARY: GLEW_SHARED_LIBRARY-NOTFOUND
-- FindGLEW: GLEW_STATIC_LIBRARY: /home/chris/work/projects/BambuStudio_dep/usr/local/lib/libGLEW.a
-- FindGLEW: GLEW_LIBRARIES: /home/chris/work/projects/BambuStudio_dep/usr/local/lib/libGLEW.a
-- FindGLEW: GLEW_VERSION_MAJOR: 2
-- FindGLEW: GLEW_VERSION_MINOR: 2
-- FindGLEW: GLEW_VERSION_MICRO: 0
-- FindGLEW: GLEW_VERSION: 2.2.0
-- FindGLEW: Creating GLEW::glew_s imported target.
-- FindGLEW: Creating GLEW::GLEW imported target.
-- Found NLopt in '/home/chris/work/projects/BambuStudio_dep/usr/local/lib'.
-- Using NLopt include directory '/home/chris/work/projects/BambuStudio_dep/usr/local/include'.
-- Using NLopt library '/home/chris/work/projects/BambuStudio_dep/usr/local/lib/libnlopt.a'.
-- Could NOT find OpenVDB (missing: OpenVDB_INCLUDE_DIR OpenVDB_LIB_COMPONENTS openvdb) (Required is at least version "5.0")
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:549 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:549 (find_package)

-- TBB found in /home/chris/work/projects/BambuStudio_dep/usr/local/lib/cmake/TBB
-- Found Boost: /home/chris/work/projects/BambuStudio_dep/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0") found components: iostreams system 
CMake Error at CMakeLists.txt:554 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

-- Configuring incomplete, errors occurred!
See also "/home/chris/work/projects/BambuStudio/build/CMakeFiles/CMakeOutput.log".
gmake: Makefile: Datei oder Verzeichnis nicht gefunden
gmake: *** Keine Regel, um „Makefile“ zu erstellen.  Schluss.
```

### Expected results

Configuration complete.

### Project file & Debug log uploads

[CMakeOutput.log](https://github.com/bambulab/BambuStudio/files/14547459/CMakeOutput.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-03-12:
Hello @Christian-Kauth 

is your building of deps successfully?
could you give a full log of the building of your deps building

vf02, 2024-03-19:
You should configure it with the following cmake arg : 
`-DOPENVDB_FIND_MODULE_PATH="/usr/lib64/cmake/OpenVDB"`

ejg3855, 2024-04-01:
I am also experiencing this issue when I see

//Path to a program.
OPENVDB_PRINT:FILEPATH=OPENVDB_PRINT-NOTFOUND

ejg3855, 2024-04-01:
CMake Deprecation Warning at cmake/modules/FindOpenVDB.cmake:126 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  CMakeLists.txt:549 (find_package)

lanewei120, 2024-04-01:
> CMake Deprecation Warning at cmake/modules/FindOpenVDB.cmake:126 (cmake_minimum_required): Compatibility with CMake < 3.5 will be removed from a future version of CMake.
> 
> Update the VERSION argument value or use a ... suffix to tell CMake that the project does not need compatibility with older versions. Call Stack (most recent call first): CMakeLists.txt:549 (find_package)
> 
> CMake Warning (dev) at cmake/modules/FindOpenVDB.cmake:160 (if): Policy CMP0057 is not set: Support new IN_LIST if() operator. Run "cmake --help-policy CMP0057" for policy details. Use the cmake_policy command to set the policy and suppress this warning.
> 
> IN_LIST will be interpreted as an operator when the policy is set to NEW. Since the policy is not set the OLD behavior will be used. Call Stack (most recent call first): CMakeLists.txt:549 (find_package) This warning is for project developers. Use -Wno-dev to suppress it.
> 
> CMake Error at cmake/modules/FindOpenVDB.cmake:160 (if): if given arguments:
> 
> ```
> "NOT" "openvdb" "IN_LIST" "_OPENVDB_COMPONENT_LIST"
> ```
> 
> Unknown arguments specified Call Stack (most recent call first): CMakeLists.txt:549 (find_package)

@ejg3855 which platform are you using?
do you follow the building guide there https://github.com/bambulab/BambuStudio/wiki?
and could you show the detailed steps here?

ejg3855, 2024-04-02:
I am following the wiki to a "T" 

I am installing on Ubuntu 23.10

Its the 3rd step after compiling the dependencies

lanewei120, 2024-04-03:
I just tried on my ubuntu 22.04
it can success

please share the full logs and commands from the first step

thanks

ejg3855, 2024-04-03:
Where can I find the log files?

lanewei120, 2024-04-06:
just the full cmake output logs from beginning

you can redirect the output logs to a file and upload it here

Xses-1, 2024-10-03:
Same issue

Xses-1, 2024-10-03:
So I did:
`sudo apt install libopendvb-*`
and it worked but now I am stuck on something else.

miroslav-beranic, 2025-01-14:
Hi, I've run into the same issue. Documentation is slim, reading: https://github.com/SoftFever/OrcaSlicer/wiki/How-to-build, where all it is written is: `BuildLinux.sh -udisr`.
I got more options from:
```
$ ./BuildLinux.sh -h
Usage: ./BuildLinux.sh [-1][-b][-c][-d][-i][-r][-s][-u]
   -1: limit builds to 1 core (where possible)
   -b: build in debug mode
   -c: force a clean build
   -d: build deps (optional)
   -h: this help output
   -i: Generate appimage (optional)
   -r: skip ram and disk checks (low ram compiling)
   -s: build orca-slicer (optional)
   -u: update and build dependencies (optional and need sudo)
For a first use, you want to 'sudo ./BuildLinux.sh -u'
   and then './BuildLinux.sh -dsi'
```
I ran `BuildLinux.sh -udisr` and it said: 0 to be installed. After that, I started to build specific: `$ ./BuildLinux.sh -csr`. Where I got:

```
-- Could NOT find OpenVDB (missing: OpenVDB_LIB_COMPONENTS openvdb) (found suitable version "8.1.0", minimum required is "5.0")
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:318 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:620 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:325 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "8" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:620 (find_package)

-- TBB found in /usr/lib/x86_64-linux-gnu/cmake/TBB
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: iostreams system 
-- find_package could not find Blosc. Using fallback blosc search...
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test LLVM_HAS_ATOMICS
-- Performing Test LLVM_HAS_ATOMICS - Success
CMake Error at CMakeLists.txt:625 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

```

Any update on this? Am I missing something?

miroslav-beranic, 2025-01-14:
I was missing dependeny:
```
apt install libopenvdb-tools
```