bambulab/BambuStudio #3699: X1C Not Printing Studio Designed GCODE as pictured in Studio

kamland, 2024-03-10:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Sonoma 14.2.1 (23C71)

### Printer

Bambu Lab X1 C

### How to reproduce

Load project file to Studio
Generate code
Send to printer and print

### Actual results

[image]
[DogInCup.heic.pdf](https://github.com/bambulab/BambuStudio/files/14550734/DogInCup.heic.pdf)

### Expected results

Should print as pictured in Studio.

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/14550724/Archive.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-03-11:
[image]
the flush volume was set to 0, need to be set to 1

EricDimitri, 2024-03-16:
> ![image](https://private-user-images.githubusercontent.com/112676417/311572800-bdc15479-229f-483f-b4b9-28c079fb23f4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Just for me to know, where is that window? I never saw it. Thanks!!

QingZhangBambu, 2024-03-18:
@EricDimitri click here
[image]

EricDimitri, 2024-03-18:
> @EricDimitri click here [...]

And how are those calculated? Because I am not using Bambu filaments (those dont reach my country) so I see in that window that volumes are different, depending on colour? Or how does it work?

QingZhangBambu, 2024-06-20:
@EricDimitri sorry to reply so late, I missed your email.
 Yes, the flush volumes depend on color. We do tests of Bambu filaments to get these data. 
you can modify the data on the flush window and watch the flush process on the printer. If the filament is clean, then the data is fine.

closed 2024-10-14 by QingZhangBambu