bambulab/BambuStudio #3702: Abnormal number of Page Faults even when the software is not being used, sitting in the background.

EricDimitri, 2024-03-11:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 pro

### Additional system information

CPU: Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz   2.81 GHz
RAM: 32 GB
GPU: Nvidia Gforce GTX 1070 GDDR5 @ 8GB (256-bit)

### Printer

Bambu Lab P1S AMS

### How to reproduce

1 - Open Bambu Studio
2 - Slice any File
3 - Move to other tab different than "Prepare" tab

(Even without any file or project opened, the problem exists)

### Actual results

Page Faults are constantly being generated, several thousand per second, even hours after not touching Bambu Studio

### Expected results

Page faults should be hundreds of times lower. Bambu studio generates more page faults than the whole system, several times.

This huge ammount of page faults affects system performance. It has to be stopped.

### Project file & Debug log uploads

It doesnt matter, it can be reproduced with any file.

### Checklist of files to include

- [X] Log file
- [X] Project file

EricDimitri, 2024-03-11:
https://github.com/bambulab/BambuStudio/assets/31411449/5449ea4c-c78a-457c-93d7-6bf208dcfd33

Check on top, sorted by pagefaults, the ammount of page faults, and the PF delta (page faults per second) insanely high. That makes system slow down to the point that scrolling on a windows explorer window for example are choppy.

lanewei120, 2024-03-12:
what is the tool used to monitor the Page faults?

EricDimitri, 2024-03-12:
> what is the tool used to monitor the Page faults?

It is Process Explorer
https://learn.microsoft.com/en-us/sysinternals/downloads/process-explorer

Microsoft official tool

oyidi, 2024-03-12:
I have same problem, every software very lagging before I close Bambu Studio.

lanewei120, 2024-03-12:
> > > I have same problem, every software very lagging before I close Bambu Studio.
> > 
> > 
> > @oyidi it may be different reason
> > could you give a snapshot of your task manager? if it is very lagging, the cpu occupy ratio and memory occupy ratio should be abnormal
> > and what is Bambu studio doing when this happens?
> 
> Seems I meet the same reason with this issue. cpu, memory, disk occupy ratio were normally, but the computer was very lag. I have post same issue(#2959) and work order(CN231220082001) last year, now I using Bambu Studio 1.8.4.51, the lagging problem seems not fixed.
> 
>  Process.Explorer.-.Sysinternals_.www.sysinternals.com.DESKTOP-FKFMM0T_x2697.2024-03-12.16-33-12.mp4

@oyidi 
现在是每次启动bambu studio都会这样吗？
有必现的步骤不？

另外出现这种情况，bambu studio是不是没在做啥？如果在里面新建一个项目能恢复么
最后如果没有做啥事情的话，cpu的占用率应该不到1%，如果是接近10%也不太正常, 
出现问题的时候最好能帮忙截个任务管理器的图

多谢

EricDimitri, 2024-03-12:
I would really appreciate you to type in English, so I can understand the whole chain. Thanks!! :)

EricDimitri, 2024-03-12:
Hi!! Do you need any extra info? Were you able to reproduce? Please keep me posted. Thank you!! :)

lanewei120, 2024-03-13:
> Hi!! Do you need any extra info? Were you able to reproduce? Please keep me posted. Thank you!! :)

@EricDImitri

I can reproduce it here following your steps
however, I don't feel lagging on my computer

is it lagging on your computer?

thanks

EricDimitri, 2024-03-13:
10, 20 thousand page faults affects the performance of every computer, you may not notice it, but it does affect, a lot, I tested in all my comnputers, and it affects, all of them, and, even if it doesnt, do you think that that awful performance can be left as is? Something is performing terribly wrong in that software, and just because you dont notice the performance decrement, do you think can be left kust because?

come on!!!

Scroll thru the files in a windows explorer window, try to do it in a 4K monitor, and tell me that it doesnt affect the scrolling ... It is terribly wrong. Please, inveswtigate and fix that.

I left it running on my computer, with other big software, and Bambu Studio generated 20 times more Page Faults than ALL THE REST OF PROGRAMS INCLUDING OS COMBINED

Please, dont say that just because you dont notice it, it is not important :)

lanewei120, 2024-03-13:
yes, we will find the reason and fix it
don't worry

EricDimitri, 2024-03-13:
Let me give you as precise as possible, is not that my computer is unusable... it is just that you notice that there is some heavy process in the background, like when you have your cpu to 90% lets say, or when you are swapping to HArd drive because you run out of mem... you "feel" it lagish, you know what I mean?

THANK YOU VERY much for taking care of this. Studio is awesome, and it is even better to see that dev team takes care of problems users have, and takle them. THANKS!!!

Whatever I might do to help you with this, just let me know. 

Do you have any idea if it will be in the plate for next release? or  any ETA?

THANKS AGAIN!!!!

EricDimitri, 2024-03-19:
Hi !! @lanewei120 @XunZhangBambu, any update on this? Will it be fixed for next release? when is that going to happen? Any ETA?

Thanks!!!

lanewei120, 2024-03-20:
@EricDimitri 
we have found some page faults with preview and device,
however the root reason is still under debugging

do you have any debug experience for such kind of issues?

thanks

EricDimitri, 2024-03-20:
> @EricDimitri we have found some page faults with preview and device, however the root reason is still under debugging
> 
> do you have any debug experience for such kind of issues?
> 
> thanks

Great you are making progress... No, unfortunately I dont. I guess it is one of those issues "hard to fix" I checked, and Orca has the same issue, you know, orca slicer. But Prusa doesnt. May be you can ask Prusa dev team if they fixed after the fork, or joint with Orca Slicer to try to figure that out. It is insane that ammount of pagefaults being idle for hours. I dont get what could be happening. May be Prusa slicer team is a good start :)

XunZhangBambu, 2024-03-21:
@EricDimitri 
Through my testing, I found that the page fault exception persists only on the device page. This is because a timer is set on the device page to refresh various status information during printing. On other pages, the page fault exception does not persist after staying for a while.

EricDimitri, 2024-03-21:
> @EricDimitri Through my testing, I found that the page fault exception persists only on the device page. This is because a timer is set on the device page to refresh various status information during printing. On other pages, the page fault exception does not persist after staying for a while.

Hi!! Weird, I have it open to test, in 2nd monitor, no interaction at all, in Preview Page, and after 3 hours it still throws more than 1000 page faults per second, and counting... I guess there must be something that Prepare page doesnt have, that all the others have, that generate pagefaults. 

when you say "after a while" how much time is that  "a while" ?

Thanks again for going thru this.

EricDimitri, 2024-04-15:
Hi Guys!!
How is it going @XunZhangBambu , @lanewei120 , @bambu123 ? DSo you have any update on this? NANy progrss worth mentioning?

Thansk a lot!!!

    Eric

lanewei120, 2024-04-16:
@EricDimitri 

on 1.9.0.70 public beta3,
we have improved the logic when bambustudio in background with device tab
and we are trying to do more when it is on the device tab on foreground also.
however, we have some other urgent task, we need to do it a bit later

for the preview page, we have some clues, but don't have much progress
we will continue to dig more

thanks

EricDimitri, 2024-04-16:
Great!! It makes Sense. Of course there are more important things than
this, but glad to hear that you are doing some progress. Thanks!!!

muireng, 2024-06-23:
I would like to add that I have the same problem, laggy performance on my computer with Bambu Studio running. I usually close it after I start my 3d print job.

k-y, 2024-07-11:
Same issue; sometimes I forget to close Studio after printing a model.  Over time, the System process will persistently exceed 10% utilization and after closing Studio this process goes back down to < 5%.  Also, while Studio was open, opening files (small or large, didnt matter) via Windows Explorer was slow; after closing Studio, back to opening files instantly.

solsystemlabs, 2024-07-13:
I can also confirm that I have this issue. Considering that Bambu Studio doesn't provide a way (like tabs) of having multiple projects open at the same time, requiring me to have multiple instances open at the same time, I quickly end up with a very jerky and laggy PC. I really like the ecosystem Bambu provides via Studio, but I'm strongly considering switching to Orca Slicer, though I really don't want to :(

nesttle, 2024-08-10:
Same here... No solution yet?

zygoman, 2024-08-15:
Wanted to weigh in with my two cents worth....
I love the functionality of my Bambu P1S, and also how Studio works in general.  It's the idle behavior that is lacking, as several folks have noted.
I have basically the same symptoms as others have noted in this thread.  Nothing happening or being done within Bambu Studio in background, and my whole PC is sluggish.  I have done some very unscientific testing, and found that indeed leaving Studio running in the background on the _Device_ tab gives the worst performance hit.  The _Prepare_ tab is much less impactful, but still contributes to enough CPU activity that my laptop fans are constantly revved up.  I hate to constantly be closing Studio and starting it up again every time I want to slice and print.
I just kind of stumbled onto this thread, and it was representing my experience exactly, so I encourage the developers to take this seriously.  BTW, I spent 30 years in IT doing server builds/support, everything from 386's to bigillion-core virtualization, so I can vouch that the stats being shown by Process Explorer are legit.  The OP is not making this up or using some DIY analysis software.  Gratuitous Page Faults (in the background) of the magnitude being shown here are the IT equivalent of chest pain that radiates into the arms; somebody needs to get to the emergency room for a complete workup!

nesttle, 2024-08-16:
I would like to add to this thread that this problem is also present in OrcaSlicer (although it would be nice if someone could corroborate the Page Faults, but I also have many Page Faults in OrcaSlicer), so it does not seem to be an error exclusive to Bambu Studio.

I mention this because it will surely be helpful to developers, since it may be that the origin of this poor performance is before the fork of these two applications (or not).

lanewei120, 2024-08-19:
we are refactoring the logic of the device tab, which will improve this page fault issue
and it will be ready in future versions

Atrob92, 2024-08-19:
Glad to see some progress being made, this issue is very apparent on my PC, and I have i9-13900k, 64 GB RAM, NVMe 4.0 SSD and RTX 4090, crazy how much these page faults can slow down a computer.

EricDimitri, 2024-08-19:
Excellent news!! Anyway, in previous releases, it was somehow corrected, not completely, but at least the ammount of page faults 
 in some tabs is not that high, not affecting that much performance of computer. Anyway, excellent news!! THANKS!!!!

hervema, 2024-08-26:
I'm glad to see that other people are also experiencing laptop performance problems when bambu studio is running :-) This is my case too, very often (latest version 1.9.3.50) The only way to get my laptop's performance back is to close bambu studio and restart my laptop. I'm not sure if the problem is the same for everyone. My laptop is powerful enough I think :-) (I7-13800h, 64GB RAM, NVMe 4.0 SSD, RTX 4060)! I'm just sharing my experience, I don't want to open a case. I hope you'll have a number of situations with other users to resolve these performance issues. It's a pity. On the other hand, it's a great product 👍__

HYHY2233, 2024-08-31:
1.9.5.51 is still like this, do you know what happened? Even if it can't be solved immediately, can you issue a notice to explain it roughly? Or how to alleviate this lag? It's really annoying to repeatedly open and close bambu studio every day.

lanewei120, 2024-09-02:
1.9.4 and 1.9.5 are just small versions, the improvement will be included in the next big version(1.10)

EricDimitri, 2024-09-16:
When is 1.10 going to be released? Do you have any idea? THANK YOU!!

lanewei120, 2024-10-02:
@EricDimitri 
we will start related work after our holiday
however, we don't have a fixed date 

thanks

EricDimitri, 2024-10-02:
@lanewei120 Thanks for the update but, I have no idea when your holidays are :)

EricDimitri, 2024-10-14:
> @lanewei120 Thanks for the update but, I have no idea when your holidays are :) Also copying @bambu123 and @walterwongbbl

Any update on this? Thank you!!!

lanewei120, 2024-10-15:
> > @lanewei120 Thanks for the update but, I have no idea when your holidays are :) Also copying @bambu123 and @walterwongbbl
> 
> Any update on this? Thank you!!!

we will release 1.10 public in a short time, thanks

EricDimitri, 2024-10-16:
> > > @lanewei120 Thanks for the update but, I have no idea when your holidays are :) Also copying @bambu123 and @walterwongbbl
> > 
> > 
> > Any update on this? Thank you!!!
> 
> we will release 1.10 public Beta in a short time, thanks

IS there any place where i can see release notes before release? or that doesnt exist? :)

THANKS A LOT!!!

lanewei120, 2024-10-17:
> IS there any place where i can see release notes before release? or that doesnt exist? :)
>
> THANKS A LOT!!!

https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74
the public beta version has been already released on Tuesday

zygoman, 2024-10-17:
Thanks for the beta-release code!
Just looked over the release notes.  Not seeing anything in the bug-fix section even remotely talking about idle performance issues or thread-hogging.  If this release includes such fixes, is there a reason that it would not be mentioned in the release notes?
I'm installing the beta release now and I'll get back to this thread on my observations.

lanewei120, 2024-10-17:
it is in the Improvements list

[image]

EricDimitri, 2024-10-17:
> it is in the Improvements list [...]

Amazing!! Thanks a lot!!!!

zygoman, 2024-10-17:
Yes, indeed. Thanks for pointing out where that came into the release notes.
I appreciate your quick response.

EricDimitri, 2024-10-17:
> https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74 the public beta version has been already released on Tuesday

I am tersting the 1.10 beta, and the performance issues look awesome!!! it is an outstanding improvement!! THANK YOU VERY MUCH!!!!

internoot, 2024-12-06:
This issue seems to be present in my installed version 1.10.1.50 also.

With a single Bambu Studio window open, once I switch to device view I also see thousands of page faults per second. As soon as I switch to another tab the page faults stop.

[image]

Performance in the device tab is quite poor, especially when I have multiple open windows (eventually it becomes unbearable and I restart bambu studio).

mmodrow, 2025-01-23:
I'm running Bambu Studio v1.10.1.50 on a powerful Windows 11 machine (AMD Ryzen 7 5800 Octocore, 32GB RAM and an NVIDIA GeForce RTX 3080).

The machine runs extremely smooth until I open Bambu Studio. Especially when switching to a different application with BS in the background (pun intended), the UI gets extremely sluggish; the mouse takes ~10s just to cross the screen and switching applications takes ten of seconds up to minutes.
It only gets worse with more BS instances in the background. Currently I am running 4 instances for multiple prints I have prepared and the computer is rendered virtually useless at this point, forcing me to write this comment on my phone, as writing text in a chrome tab would take tens of seconds per word...

lanewei120, 2025-01-23:
> I'm running Bambu Studio v1.10.1.50 on a powerful Windows 11 machine (AMD Ryzen 7 5800 Octocore, 32GB RAM and an NVIDIA GeForce RTX 3080).
>
> The machine runs extremely smooth until I open Bambu [...]

@mmodrow 
are you also leaving bambustudio on the device page?
could you give a short video on your cases?

mmodrow, 2025-01-23:
I have closed all but three instances, that are all on the preview tab with readily sliced projects.
Besides this there is a single chrome instance running. All the mouse movements in the attached video were VERY snappy inputs as in "move the mouse for 1/3 of a second, wait for execution, repeat moment pattern. 
https://github.com/user-attachments/assets/76f98d28-4c96-40cf-aeda-d20a862e7e9a

internoot, 2025-01-27:
After encountering this issue I switched to running bambu studio under linux where it doesn't occur.

I just tried again under windows on a fast i7 system with 32gb RAM and accelerated graphics, but the performance is significantly worse than on my low powered NUC running ubuntu.

lanewei120, 2025-01-30:
@internoot @mmodrow 
are you using two graphics card?
if yes, could you show your two cards here

internoot, 2025-01-30:
@lanewei120 just one graphics card (geforce 3060ti) in my case (but it is connected to 3x 4k displays).

mmodrow, 2025-01-30:
> @internoot @mmodrow 
> are you using two graphics card?
> if yes, could you show your two cards here

Nope. The internal graphics of the ryzen are not in use, if there are any. Just the GeForce rtx.

internoot, 2025-02-15:
@lanewei120 any update on this? i have basically given up using bambu studio under windows completely.

lanewei120, 2025-02-15:
hello @internoot 

we have tried to find a computer to reproduce this issue, but still failed

currently we have done following things:
1. found more page faults on the device pages and improved it internally
however, this is not the root cause of your case
2. tried lots of computers, but can not reproduce
however, we found a computer have a similiar behavor
after opened 3 bambu studio running more than 24 hours

are you always running 3 bambu studios at the same time?
and if only run one bambu studio, is this issue still there?

thanks

internoot, 2025-02-15:
Unfortunately even a single instance of bambu studio impacts performance on my windows computers. If I open 3 instances, I will quickly close them in order to be able to continue to use the computer.

Under linux I often have 3+ instances of bambu studio running for weeks at a time without any issue.

internoot, 2025-02-15:
@EricDimitri are you still experiencing this issue?

EricDimitri, 2025-02-15:
> [@EricDimitri](https://github.com/EricDimitri) are you still experiencing this issue?

Hi!!! I changed compuiter for an even more powerful computer. I dont know if it is related or not, but may be it is related to nvidia graphic cards? check if it is something that matches. I mean, if everyone having the problem has an Nvidia Graphics. Just my 2 cents. In me new set up, I do have one, but it is a different one, and The computer is much much better so, ....

digregoriovalerio, 2025-12-27:
> > BAMBU STUDIO is a shit software.I have this problem too.
> > This problem is not solved on 2.4.0.70
> 
> Don't use it if you think so. Pretty simple.

What should people use if they already bought a Bambulab?

digregoriovalerio, 2025-12-28:
And tell me, if we use orca slicer what is going to be missed? Bugs apart... Your comments are pointless. It's evident Bambu studio is unusable on win 11 on laptop with Nvidia card and dual screen. I found already tens of people with the same issue. I found a guy that solve the issue by detaching the HDMI of second screen. The problem isn't the user.

EricDimitri, 2025-12-28:
The slicer, Bambu Studio, is given FOR FREE... and you can use any slicer you want... IT HAS NOTHING TO DO WITH THE PRINTER... Come on man... Useless? Please...  Use ORCA and thats it, if it is that wrong for you... More than a dev, you sound like a KAren ;)

bfrancom22, 2026-01-19:
Yep I have the same issue. On two separate high ends PCs that share zero config similarities aside from OS build. Both are fully independent systems with different hardware.

[3 more replies]