bambulab/BambuStudio #3703: support switching to any backup filament loaded in the AMS, regardless of type/color

tsdexter, 2024-03-11:
**Is your feature request related to a problem? Please describe.**
You cannot change to another filament of a different color/type after a runout

**Describe the solution you'd like**
Allow changing the filament after a runout to any other slot regardless of color or type. If the filament is not an exact match, show a warning saying something about the filament differences causing potential issues such as failed print, nozzle clog, etc. 

The user should be able to decide on their own whether or not they want a multicolored print or a potential nozzle clog. Most of the time, we just want to switch to another color because we don't have more of the original color and it's better than throwing out half a print. In some cases, like switching from Bambu PLA to eSUN PLA we can determine that they print nearly identical and there isn't going to be an issue, so we should be able to switch to it. 

**Describe alternatives you've considered**
Manually moving the new filament to the same slot and saying it's the same type/color (ie: if we're able to do this anyway, then what is the issue allowing us to just do it from software)

**Additional context**
this is discussed in #1256 which is closed but is still an issue - I've created a new issue assuming you aren't paying much attention to new comments on closed issues.  See: 
[image]

seanbetts, 2024-03-11:
I'd love to have this feature too.

guillaumemolter, 2024-03-11:
@tsdexter thank you for opening a new issue.

tsdexter, 2024-03-11:
@guillaumemolter no problem - after 6 different requests to reopen and there is not even a comment/reply from the team, I figure a new one will get more attention

eried, 2024-03-12:
yes please, just allow resuming to ANY spool

picklefactory, 2024-03-19:
This would be fantastic for running out the last few grams of filament on a spool on a "I don't care what color it is" print without having to take a trip to the printer for each one.

s-pankhurst, 2024-03-27:
only had my P1S for a week and twice now I've wished this was a standard feature.

mb323, 2024-04-05:
+1 for this feature.

Had 4 spools of Bambu Basic PLA, when one ran out, I had to physically move another spool to the empty spool slot to get it to continue. There was no option for me to pick another spool to use.

I agree with the "I don't care what color it is" option but should be able to select another color if wanted at the time the runout happens.

bordermultimedia, 2024-05-07:
I'm currently having issues with one slot of the AMS (not being able to load/unload correctly), and it would be a life saver to be able to change slots manually mid print. I will have to unplug the AMS, repair it, and plug it again to resume the print, that is a pain and a waste of time. Please consider being able to change filament type/colour of the AMS while paused.

ChrisVonFrosta, 2024-06-19:
PUSH
Bring it, pls.

mike-levenick, 2024-06-20:
+1. New to bambu, and I cannot believe this isn't an option. Feels like a no-brainer.

tmrt, 2024-06-20:
+1 would be extremely helpful

Jordan-Morales, 2024-06-28:
👍 
This is an important feature, continuing a print immediately is much better for integrity, and if color doesn't matter it should be supported.

Raiden38, 2024-07-04:
Yes PLEASE!!! Just did a 6h print and my white spool run out, not being able to complete the 2 remaining layer of white!!! At that point, I wouldn't mind just ignoring the remaining white and just continue with the blue spool that is 95% of the part!

Alg3braic, 2024-07-18:
+1 This is a needed feature, put it in an advanced tab or something but it really shouldn't be withheld. My printer will be idling hot till I get home and swap the filament to a different color anyway.

tomasz90, 2024-07-27:
LETS DO IT Bambu, everyone is waiting for this....

eried, 2024-07-29:
it is so stupid :/ sometimes is better not to have genuine Bambu spools for this issue

jfxg, 2024-08-08:
+1 I've had my AMS for a few weeks now and I'm running into this issue with multiple spools running low. It's really annoying when you're printing dozens of storage bins / grids and don't care about the color. I don't have this issue on my Prusa Mk4 with MMU, that has a dead-simple "Spool Join" option to tell it "When Filament 1 runs out, continue with Filament 2, then 3..." It is a little baffling that Bambu doesn't have a similar feature even 18 months after the original issue was opened.

karcaw, 2024-08-08:
> +1 I've had my AMS for a few weeks now and I'm running into this issue with multiple spools running low. It's really annoying when you're printing dozens of storage bins / grids and don't care about [...]

If you set all your spools to the same color, and type, the AMS will fail over to the other spools.  now if it auto-picks the colors based on the RFID it kinda screws this up.

sucho92, 2024-08-11:
+1 this would help immensely

eried, 2024-08-12:
> If you set all your spools to the same color, and type, the AMS will fail over to the other spools. now if it auto-picks the colors based on the RFID it kinda screws this up.

RFID is clearly the problem here, otherwise is easy just to put any color to continue.

jfxg, 2024-08-12:
> > If you set all your spools to the same color, and type, the AMS will fail over to the other spools. now if it auto-picks the colors based on the RFID it kinda screws this up.
> 
> RFID is clearly the problem here, otherwise is easy just to put any color to continue.

I wouldn’t say RFID is the problem. Yeah the RFID gets in the way of the workaround, but the problem is that we need a workaround in the first place. I 100% _want_ the RFID feature, it’s fantastic that I can drop a spool in the AMS and have it know what color/type it is automatically and be able to see that in Bambu Studio. Giving that up to allow it to auto-continue prints with alternate filaments is not ideal. 

The problem isn’t the RFID, the problem is that the auto-switch only works for identical colors/brands.

Alg3braic, 2024-08-12:
> RFID is clearly the problem here, otherwise is easy just to put any color to continue.

Naw, because you can set your colors to anything you'd like before printing regardless, of rfid. The rfid just allows for auto syncing of ams loadout. The problem is 100% not having a software option for a common user inconvenience.

ATunaMelt, 2024-10-02:
Please add this feature. When parts are purely functional and color is not a factor, we need the ability to continue the print regardless of color/type. I don’t want to waste the filament I’ve already used to get a print to 80% done….

ChrisVonFrosta, 2024-10-03:
It's hard to believe that the company that built this great printer hasn't already added this must-have feature. The necessary effort is hardly in proportion to the negative impression it creates.

tsdexter, 2024-10-04:
any updates @YifanWuBambu ?

mb323, 2024-10-07:
What's happening with this update? Seems the auto refill feature may be broken in the latest version. I just had an 8 hour print stop in the middle and refuse to load the identical Bambu filament from another AMS slot. I had to move it to the slot where the spool was that ran out before it would load and continue. This is the first time Auto Refill has failed.

bartcerneels, 2024-10-11:
Adding my +1 to this feature request.

Ran into this today: a matte grey spool ran out on the last 5% of a functional print and I have matte black in another AMS slot. I'm out for the night so would like to continue the print remotely using the matte black but there is no option to force it.
Please give us this choice Bambu. It's a huge benefit of the AMS and your ecosystem with the handy app and connected firmware.

chwdy, 2024-10-21:
+1 to this feature request.
will be really helpful to use left over filament for drafting

Jokoe89, 2024-11-16:
Another +1, for parts that are not color dependent it would be super useful to, after manual confirmation that you accept different colors, allow it to switch to other spools.

scuba75, 2024-12-11:
weirdly my just did this.
switched from black that ran out to blue, didn't even ask me.
all bambu filament with RFID's

eried, 2024-12-12:
> weirdly my just did this. switched from black that ran out to blue, didn't even ask me. all bambu filament with RFID's

Damn, black to blue? did they implemented this already and we are just complaining for nothing :P I need to test next time, I pretty much avoid printing with bambu filament with RFID long objects due this issue, in case I dont have the exact same one to continue

Freshmn09, 2025-01-07:
Industry Prints do not care 99% of the time if they are multi coloured or not. I need the option to overide colour on request to keep printing had prints sit for 5 days waiting on filament before.

MokretsovD, 2025-01-07:
Dear Babmu team, 

please fix the issue or, at the very least, be respectful, and reply to all those people who bought your XXXXEUR/USD machines + AMS so they can solve runout events the exact same way they would do it on any 3d printer for 200$ - by manually inserting new spool.

Are you aware of the significant issue described here? Is this feature on your roadmap? Can we expect this to be fixed in Q1 (Q2? Q3?) 2025? Why has there been no reply from anyone responsible for fixing this for a year?

BTW, I am almost sure, that this issue harms the sales of your own filament as well.

This feature is highly requested, please, have a look at it.

cpomerantz, 2025-01-24:
I don't understand how difficult it would be to allow a filament change/selection to be done while paused. I've had the printer a little over a month now and have wanted this about 3 times. Once because I had a model that put the pause in there and I didn't know you could not do a manual switch, the others were from filaments running out.

Kevegl, 2025-02-23:
+1 to this feature request.
This would be really useful!

dsperman, 2025-04-16:
Another request for this feature.  Please make it happen.

eried, 2025-04-20:
Not sure anyone from bambu actually reads suggestions

callmestech, 2025-07-05:
Please implement this feature. This is very upsetting that I need to go home to put the same filament from another filament slot to continue printing. Is this what we deserve ?

tsdexter, 2025-07-07:
> Please implement this feature. This is very upsetting that I need to go home to put the same filament from another filament slot to continue printing. Is this what we deserve ?

they shouldn't even need to read this, it should be common sense to have this

revspete, 2025-07-30:
+1 for this request

Also facing the same problem, half way through a 9 hour remote print.  I have the same Bambu Lab material in different colours loaded in the other slots but not able to select them.  Does not feel great to leave the printer sitting warm for a couple days until I can physically change the filament.  What a waste of filament.

caprageon, 2025-08-17:
+1 to this request.
This has been requested for years now !

tsdexter, 2025-08-22:
> +1 to this request. This has been requested for years now !

the original issue is only 6 months older than the release of the original X1C

eried, 2025-08-26:
everyone with the same text lol 😂

```
Hello Bambu team,

When filament runs out, the printer only allows reloading the same slot. This makes prints stop and waste time. Please allow users to choose any spool in the AMS after runout. If type/color is different, just show a warning.

[Issue #3703](https://github.com/bambulab/BambuStudio/issues/3703)

Thank you,
[Your Name]
```

dyburke, 2025-09-28:
Unbelievable this isn't an option... Also what's with the ignore on the ticket? 2+ years? The core of the printer is ease of use; the core of the AMS is easy filament management/switching.  Seems like a small lift for a large benefit given there's already adjacent features.

pkathro, 2025-11-04:
Wow, I just bought my first Bambulab printer and cannot believe that a) this isn't already a feature and b) they are ignoring so many requests to fix this shortcoming. 
I'm starting to wonder whether I should just send it back for a refund and go with a company who cares about and listens to their customers whilst I'm still within the returns window :(

perhusaas, 2026-01-17:
+1 to get this feature ASAP. 
I'm mostly printing functional stuff, and colors is way less important than having to cancel a long print because i was too optimistic on filament usage.  We should be able to select a different slot to complete the print, or at least be able to edit/reload the AMS while the printer is paused?
This would offcourse leave the responsibility of using equivalent filament to the user, so perhaps this should have a disclaimer when you enable it. But it should be possible!
@StoneLiBambulab?

ChrisVonFrosta, 2026-01-29:
One reason why other manufacturers will get my attention.

This is unbelivable

cpomerantz, 2026-02-27:
Yeah, honestly, wtf. This is such a simple thing. I can't believe it is any more difficult than a few lines of code to implment.

callidex, 2026-10-11:
In Handy just pop up the filament list and let the user choose. Pausing a huge print initiated remotely and needing physical access when there are 3 other full AMS units is insane

atomarch, 2026-11-01:
Another +1

mmarradi, 2026-11-06:
+1
Yes pleeeeease.