bambulab/BambuStudio #3704: can't printing at the same speed

Anton1111223131, 2024-03-11:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Built from source by yourself or others

### OS version

windows 10

### Additional system information

fx-8100 16gb ram

### Printer

bambu lab p1s

### How to reproduce

I can't set up printing at the same speed. the printing speed varies on different layers and on the cover of the model the speed drops to 20mm. 
How can I set the constant same speed in the settings for the entire printing time?

### Actual results

I can't set up printing at the same speed. the printing speed varies on different layers and on the cover of the model the speed drops to 20mm. 
How can I set the constant same speed in the settings for the entire printing time?

### Expected results

Я хочу чтобы принтер печатал на одинаковой скорости все слои и не было блеска на детали от разных скоростей печати.
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/14560564/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

EricDimitri, 2024-03-11:
The actual speed depends on several factors. Probably, what is happening to you is that, the minimum layer time is reached, so the speed is lowered to keep that number, and not print a layer to less speed. You should go and change that number to a lower value, and you will see that it will mantain the speed.
[image]
There you have how to change that. Anyway, when your layer is so slow, that will be printed in less than that time, it is ok to lower the speed, otherwise the filament will not have enough time to cool down, and you will get bad quality print.

EricDimitri, 2024-03-13:
Adding to that, if the problem is that you cannot reach the speed you set in the slicer, and not because of the reason previously stated, will probably be because of the MAX VOLUMETRIC SPEED set of the filament. It is in the same place of my screenshot, but in the FILAMENT tab, at the bottom. The value there for generic filament is way too conservative. You can increase that volume a lot. It depends on the specific filament, how much you can enlarge that number, but I succesfuly printed with volumetric speeds of 24 with all the filaments I tested. Keep in mind that you have to test and see, you have to probably increase the temp for the filament to flow better also. Keep in mind that the theorical max volumetric speed of the P1S hot end is 30 mm3/s. You should check the max of your printer if different. Hope this helps!!

closed 2024-04-22 by QingZhangBambu